signal11/hidapi #209: Downloads are double named e.g. hidapi-hidapi-0.8.0-rc1.tar.gz

barjac, 2015-01-16:
...as is the top level source dir after extraction.

amullins83, 2015-01-16:
When you download the code instead of cloning the repo with `git clone`, the branch or tag name associated with download is appended to the zip file name. If you had downloaded from the `master` branch, it would have been named `hidapi-master`. It is a little redundant that the tags include the repo name, but since that's been this repo's convention for several versions, I don't imagine that will change anytime soon.

barjac, 2015-01-18:
OK I can live with it - this happens in another project that I package so I will have to adopt the same approach with this.
Closing then.

closed 2015-01-18 by barjac