signal11/hidapi #210: Give access to the file descriptors / device handles

matlo, 2015-01-17:
First off, thanks for developing hidapi, it's very convenient to use.

The only thing I am missing is a proper access to the file descriptors / device handles.
This would allow to use poll/select or MsgWaitForMultipleObjects functions.
I know I can cast a hid_device pointer to an int or handle pointer, but it's terribly hackish and some day the file descriptor and device handles may not be aligned at the beginning of the hid_device structure anymore.

Please consider adding platform-specific functions that give access to the file descriptors and device handles.

matlo, 2015-01-19:
One way to properly work-around this issue in Linux is to look over /proc/getpid()/fd/* and use readlink() to find out which file descriptor matches to the device path.

amullins83, 2015-01-19:
I don't think platform-specific functions belong in the API. It seems like one of the major benefits of the project is abstracting out the platform differences for cross-platform applications. I could maybe see adding a function like `void *hid_get_device_handle(hid_device *)`, but I think casting the `hid_device` pointer as a device handle is the best way to go. I can't imagine changing the underlying structure in a way that would break that behavior.

signal11, 2015-01-19:
> Please consider adding platform-specific functions that give access to the file descriptors and device handles.

I'm not opposed to this functionality. It could enable the user to select() or poll() a HID device along with other file descriptors. libusb has similar. It will have to be developed differently on every platform.

If you write it up, I'll take the patch. Yes, I'm currently behind on patch merging :(

Alan.

matlo, 2015-10-01:
I ended up writing my own HID I/O abstraction layer: https://github.com/matlo/GIMX/tree/master/shared/async
It allows non-blocking I/O in GNU/Linux and Windows.
It has almost the same interface as hidapi, and it provides a function to register read/write/close callbacks.
It probably has many limitations compared to hidapi. For instance it does not fall-back to using the control endpoint if there's no out endpoint.

closed 2015-10-01 by matlo