signal11/hidapi #211: macos - Using IOHIDManagerRegisterDeviceRemovalCallback does not call perform_signal_callback() when calling hid_close()

HoriaGoran, 2015-01-19:
I am using 
IOHIDManagerRegisterDeviceMatchingCallback and
IOHIDManagerRegisterDeviceRemovalCallback
because I need to listen when my device is attached or detached from the USB port and perform an action.

I have noticed 2 things for when I close / remove the device:
1) if I run a script which sends a SIGUSR1 signal to my daemon, this will call a method from the daemon which will call hid_close() on the hid_device.
The handler function registered with IOHIDManagerRegisterDeviceRemovalCallback is not called and everything works fine (perform_signal_callback() is called, read_thread and hid_close finish as expected),

but 2)
if I pull the device from the USB port, IOHIDManagerRegisterDeviceRemovalCallback will trigger my registered handler function. In it I'm calling hid_close on the device. 
Now perform_signal_callback() is NOT called and also hid_device_removal_callback() is not called.
Now read_thread does not finish and hid_close() gets stuck in 
pthread_barrier_wait(&dev->shutdown_barrier);

signal11, 2015-01-21:
HIDAPI uses IOHIDManagerRegisterDeviceRemovalCallback() internally. You're overwriting my callback.

HoriaGoran, 2015-01-21:
HIDAPI uses IOHIDDeviceRegisterRemovalCallback not IOHIDManagerRegisterDeviceRemovalCallback. Not sure what the diference is between the two, the documentation describes both as "HID device removal callback routine" https://developer.apple.com/library/mac/technotes/tn2187/_index.html
Anyway I was able to find a workaround. 
I'm not using anymore IOHIDManagerRegisterDeviceRemovalCallback, now the handler fuction - hid_device_removal_callback() - registered internally with IOHIDDeviceRegisterRemovalCallback is called to break from the loop in read_thread(). 
In hid_device_removal_callback() through a function pointer I'm calling a method from my daemon to release the hid_device via hid_close() and the rest of the associated resources, but there is a catch:
read_thread() will not be signaled to finish until hid_device_removal_callback() returns so in the callback method called by the function pointer I have to create a new thread to do the memory releasing there and let hid_device_removal_callback() return else I again end up in scenario 2) from my original post.
I've tried everything and spent couple of days on this issue and this is the only way I've found to make it work and it does, but I wish it were a more cleaner way of doing things.
Also I've seen this other thread: https://github.com/signal11/hidapi/issues/117 that seems to be related, but no solution has been found in it.

signal11, 2015-01-21:
> HIDAPI uses IOHIDDeviceRegisterRemovalCallback not IOHIDManagerRegisterDeviceRemovalCallback.

I think it's probably used both at some point. The Apple APIs are horrible. The implementations have changed in API-breaking ways and TN 2187 is out of date.

> I have to create a new thread to do the memory releasing there

Yikes.

What I typically tell people is to just call hid_read(). If it returns -1, the device has been removed.

HoriaGoran, 2015-01-22:
I made sure that the memory release its done last and that once freed its not used by other threads.