signal11/hidapi #217: Linux hidraw: Occasional 'permission denied' when opening device

mrpippy, 2015-04-01:
My app runs `hid_enumerate()` every 500 ms, and calls `hid_open()` when a new device is enumerated.

About 1 out of 10 times, `hid_open()` fails in `hid_open_path()`, because `open()` fails with `permission denied`

It looks like this is a race condition, trying to open `/dev/hidraw0` before udev has properly set permissions on it. I put in a `system()` call to do `ls -l <path>` after `open()` fails, and this was the result:
```
hid_open_path() open() /dev/hidraw0 failed: Permission denied
crw------- 1 root root 248, 0 Apr  1 11:59 /dev/hidraw0
```
And even that result was not consistent, for several failures `ls` showed the correct final permissions.

Is there any solution to this, other than putting a few hundred ms delay in between `hid_enumerate()` and `hid_open()`?

udev rule (in `/etc/udev/rules.d`)
`ATTRS{idVendor}=="2057", ATTRS{idProduct}=="0001",MODE="0666"`

Ubuntu 14.04 32-bit running kernel 3.13.0-48-generic

amullins83, 2015-04-01:
Does it only happen when enumerating twice per second? Does the percentage of failure depend on the delay between successive enumerations? In other words, if you extend the delay to 1s, does it fail 1 out of 20 times or is it still 1 out of 10? I'm thinking there should be a way to subscribe to udev events instead of constantly calling `hid_enumerate`. It would ideally involve a new API function, but hardware events are really different between operating systems.

mrpippy, 2015-04-07:
With the delay at 1 second the failure percentage goes way down. But with the delay at 100 ms, it happened more often than at 500 ms.

Here's what I think is happening:
```
udev         | --- device detect ------------------------------------------------- set permissions on device node ----- |
hidapi/app   | ---------------------- hid_enumerate() ------- hid_open() fails ------------- |
```
I think it fails more often when `hid_enumerate()` is called more often, just because it increases the probability of `hid_enumerate()` getting called in that small window.

The root issue I see is that a device can be enumerated before the device node is actually ready to be opened. I see a `libudev` function called `udev_device_get_is_initialized()` which is described as
`Check if udev has already handled the device and has set up device node permissions and context`

I added a check with this call into `hid_enumerate()` and sure enough, as long as it returns 1 (device initialized), `hid_open()` never fails. I'll open a pull request for this tomorrow.