signal11/hidapi #221: Windows: hid_enumerate returns null when one of the connected devices is disconnected

ptkatch, 2015-04-09:
It's somewhat similar to https://github.com/signal11/hidapi/issues/102.

Symptoms:
- plug-in some USB devices (in my case - mouse, keyboard, CC reader, touch controller)
- run hidtest first time - it enumerates all four devices correctly
- plug-in one more device (Magtek EMV)
- run hidtest - enumerates all all five devices
- unplug one device (Magtek EMV) 
- run hidtest - returns zero devices
- reboot PC
- run hidtest - returns zero devices
- re-plug Magtek EMV
- run hidtest - returns five devices

It's not easily reproducible - I can't recreate on my Windows 7 Pro 64-bit, but can on Windows 7 Enterprise 64-bit, 32-bit POSReady 7 (Windows 7 Embedded), 32-bit POSReady 2009  (XP Embedded). Mouse and keyboard don't seem to cause issue, only Magtek.
Issue is present in hidapi.dll both from pre-built zip and built from Git.

Below is hidtest output on MS Windows POSReady 2009, Version 2.0, Service Pack 3

*** Magtek is unplugged ***

D:\hidapi>hidtest
unable to open device

*** Plug-in Magtek ***

D:\hidapi>hidtest
Device Found
  type: 046d c05a
  path: \\?\hid#vid_046d&pid_c05a#6&327afb95&0&0000#{4d1e55b2-f16f-11cf-88cb-001
111000030}
  serial_number:
  Manufacturer: Logitech
  Product:      USB Optical Mouse
  Release:      5400
  Interface:    -1

Device Found
  type: 0801 3004
  path: \\?\hid#vid_0801&pid_3004#7&2f6bc5fb&0&0000#{4d1e55b2-f16f-11cf-88cb-001
111000030}
  serial_number: 98C898A90508070D
  Manufacturer: MagTek, Inc.
  Product:      IPAD PLUS EMV
  Release:      6
  Interface:    -1

Device Found
  type: 2911 1500
  path: \\?\hid#vid_2911&pid_1500#6&33c3f010&0&0000#{4d1e55b2-f16f-11cf-88cb-001
111000030}
  serial_number:
  Manufacturer:   Product:      Magnetic Stripe Card Reader
  Release:      540
  Interface:    -1

Device Found
  type: 2911 16a0
  path: \\?\hid#vid_2911&pid_16a0#6&1e4a343c&0&0000#{4d1e55b2-f16f-11cf-88cb-001
111000030}
  serial_number: 0000000001
  Manufacturer:   Product:      USB TOUCH V39N
  Release:      0
  Interface:    -1

Device Found
  type: 413c 2107
  path: \\?\hid#vid_413c&pid_2107#7&2afb0898&0&0000#{4d1e55b2-f16f-11cf-88cb-001
111000030}
  serial_number:
  Manufacturer: Dell
  Product:      Dell USB Entry Keyboard
  Release:      115
  Interface:    -1

unable to open device

D:\hidapi>

***  Unplug Magtek ***

D:\hidapi>hidtest
unable to open device

DJChloe, 2015-11-17:
A long time ago I developed a piece of software in windows for a Belkin left handle gaming device. It is still on sourceforge with sourcecode (wich is in Delphi ). The project was called Martini something
The all detection chain and handling of hid devices is encapsulated in a reusable class and is very reliable. It is under gpl3 licence and the code can be reworked in C++.
If you are interested you can download the source code,  study it and reuse for Mixxx. 
http://sourceforge.net/projects/nostromon50onvi/

DJChloe, 2015-11-17:
Direct link to source code http://sourceforge.net/projects/nostromon50onvi/files/NOVA%20Suite/NOVA%20Suite%20v1.2.0.0%20final/Nova_suite_v1200_src.zip

DJChloe, 2015-11-17:
If I remember well,  hid_enumerate had to be called twice (once for initialization,  once to poll the device ) in order to be reliable

DJChloe, 2015-11-17:
It was important to make the software "hot plug and play " friendly