signal11/hidapi #222: Windows7: CRC-Error at hid_get_feature_report

Juckbart, 2015-04-20:
A want to read a state of my own HID Device with the HID API Function  hid_get_feature_report.
In the report I transfer 2 bytes of data. 
This is the context I use this function:
#define FEATURE_READY_SIZE 2
bool DeviceDriver::getReadyCommand()
{
    if (!m_handle) return false;

    unsigned char buffer[FEATURE_READY_SIZE];
    int res = hid_get_feature_report(m_handle, buffer, FEATURE_READY_SIZE);
    if (res < 0 ) {
        m_error = true;
        printf("Error at getReadyCommand (res==%d): %ls\n", res, hid_error(m_handle));
        return false;
    }
    if (buffer[1] == READY_TO_CAPTURE_CMD)
        return true;
    return false;
}
After this function i get always the message:
"Error at getReadyCommand (res==-1): Datenfehler (CRC-Prüfung)"
that there was an Error at the CRC Check.
I dont know why this happen. Do I use the API the wrong way or is this a bug? 

My device uses this HID report descriptor:
const HIDReport_Descriptor_t HIDReport_Descriptor [] =
{
	HID_UsagePageVendor(0x00),
	HID_Usage(0x01),			//vendor Usage 1
	HID_Collection(HID_Application),
		HID_LogicalMin(0),
		HID_LogicalMaxS(255),
		HID_ReportSize(8),		//8 bits per field for all ReportIDs

		                       //INPUT REPORT
		HID_Usage(0x01),		//vendor defined
		HID_ReportCount(17),	//17 fields
		HID_Input(HID_Data | HID_Variable | HID_Absolute),
		                       //OUTPUT REPORT
		HID_Usage(0x01),		//vendor defined
		HID_ReportCount(13),	//13 fields
		HID_Output(HID_Data | HID_Variable | HID_Absolute),
		                      //FEATURE REPORT
		HID_Usage(0x01),		//vendor defined
		HID_ReportCount(6),		//6 fields
		HID_Feature(HID_Data | HID_Variable | HID_Absolute),

	HID_EndCollection
};

signal11, 2015-04-20:
You didn't set the report ID before hid_get_feature_report(). See the 
docs for that function.

Juckbart, 2015-04-23:
Thank you for your help. 
After reading the docs twice I got it. Its a bit strange to send the ReportId in the same Buffer as I late expect to be the received data in it. 
Also got a second error: My Report was two Bytes large, but my report descriptor says feature reports has to be 6 byte.

closed 2015-04-23 by Juckbart