signal11/hidapi #227: Written data ends up in the read buffer

vitaminwater, 2015-07-15:
I have a Go program that does fairly basic read and write operations, not even in threads, the weird thing is, when I write something, it is well written to the device (its behaves accordingly), but then the next read has the just written data randomly placed in its buffer.

example:

I send:
`02:12:3c:20:11:00:4a:4e:03:89:00:00:00:00:00:00:00:00:00:15`

And the next read is:

`e7:e7:00:00:02:29:7e:2f:00:02:23:00:9a:3c:20:24:00:e6:d5:da:33:63:f9:f8:3e:b5:74:3b:3f:dd:bc:f0:3e:df:1a:a2:bd:e2:e5:00:43:76:02:12:3c:20:11:00:4a:4e:03:89:00:00:00:00:00:00:00:00:00:15:e7:e7:00:50`

(written packet start at the 24th position from end)

I'm not sure whether it comes from my code, the golang binding (which doesn't do much more than simple binding), or if it is a known behaviour of a wrongly usage of hidapi.

looks like issue #130, but I doubt it is it.
For infos, I'm using https://github.com/GeertJohan/go.hid for the golang binding.

Thanks

signal11, 2015-07-16:
I'm pretty sure that's not coming from HIDAPI. I don't know how that would even be possible unless it was coming from Windows that way. Check your device to see if its firmware is correct. See if you can duplicate it with the C test app (modify the C test app to read/write your device).

vitaminwater, 2015-07-20:
Yes you are right, it ended up being a bug in the device's firmware.

Thanks for the great work.