signal11/hidapi #230: hid_enumerate missing device

ran-z, 2015-08-04:
Hi,
I'm trying to use hidapi to access my MSI GS60's keyboard (Vendor ID 0x1770, product ID 0xff00).
(OS: Arch linux)
I haven't been able to access the device via "hid_open", and when I'm trying to use the "hid_enumerate(0,0)" API call (when running with sudo permissions) I get back NULL.

However I can see the device via lsusb -

sudo lsusb:

Bus 004 Device 003: ID 1770:ff00  
Bus 004 Device 002: ID 8087:8000 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 0cf3:3004 Atheros Communications, Inc. AR3012 Bluetooth 4.0
Bus 003 Device 002: ID 8087:8008 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

sudo lsusb -v (only the keyboard device's section):

Bus 004 Device 003: ID 1770:ff00  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1770 
  idProduct          0xff00 
  bcdDevice            1.10
  iManufacturer           1 MSI EPF USB
  iProduct                1 MSI EPF USB
  iSerial                 1 MSI EPF USB
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      57
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
can't get debug descriptor: Resource temporarily unavailable
cannot read device status, Resource temporarily unavailable (11)

Any clue why I can't access the keyboard device via hid_open?
thanks

amullins83, 2015-08-04:
Is the keyboard currently in use by the operating system? If so, it won't be enumerated.

ran-z, 2015-08-04:
On what level exactly do I need to turn it off then?
I tried using a simple script which turns the keyboard off using "xinput" before trying to enumerate the devices, but I got the same result.
any tips on how I might be able to turn it temporarily off so it might work?
thank you again