signal11/hidapi #234: Work on Windows 10 always returns -1

Max99, 2015-08-31:
using hid_read_timeout read data on windows 10, it always return -1.
The library has tested on Windows 10?

signal11, 2015-08-31:
Not sure. I haven't run it on 10. See if you can work out what the issue is. Does it work on 7 for you?

Max99, 2015-09-01:
Work on Windows 7, it's ok.
After debugging found equipment input report length is zero, but nonzero on Windows 7.
HidP_GetCaps(pp_data, &caps) caps.InputReportByteLength=0

The following is hid report descriptor:
Usage Page (Vendor-Defined 1)	06 00 FF 
Usage (Undefined)	09 00 
Collection (Application)	A1 01 
    Report ID (253)	85 FD 
    Usage (Vendor-Defined 1)	09 01 
    Usage (Vendor-Defined 2)	09 02 
    Usage (Vendor-Defined 3)	09 03 
    Usage (Vendor-Defined 4)	09 04 
    Usage (Vendor-Defined 5)	09 05 
    Usage (Vendor-Defined 6)	09 06 
    Logical Minimum (0)	15 00 
    Logical Maximum (-1)	25 FF 
    Report Size (8)	75 08 
    Report Count (6)	95 06 
    Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit)	81 02 
    Report ID (254)	85 FE 
    Usage (Vendor-Defined 1)	09 01 
    Usage (Vendor-Defined 2)	09 02 
    Usage (Vendor-Defined 3)	09 03 
    Usage (Vendor-Defined 4)	09 04 
    Logical Minimum (0)	15 00 
    Logical Maximum (-1)	25 FF 
    Report Size (8)	75 08 
    Report Count (4)	95 04 
    Feature (Data,Var,Abs,NWrp,Lin,Pref,NNul,NVol,Bit)	B1 02 
    Report ID (136)	85 88 
    Logical Minimum (0)	15 00 
    Logical Maximum (-1)	25 FF 
    Usage Minimum (Vendor-Defined 1)	19 01 
    Usage Maximum (Vendor-Defined 63)	29 3F 
    Report Size (8)	75 08 
    Report Count (63)	95 3F 
    Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit)	81 02 
    Usage Minimum (Vendor-Defined 1)	19 01 
    Usage Maximum (Vendor-Defined 63)	29 3F 
    Output (Data,Var,Abs,NWrp,Lin,Pref,NNul,NVol,Bit)	91 02 
End Collection	C0 
Usage Page (Generic Desktop)	05 01 
Usage (Mouse)	09 02 
Collection (Application)	A1 01 
    Usage (Pointer)	09 01 
    Collection (Physical)	A1 00 
        Report ID (4)	85 04 
        Usage Page (Button)	05 09 
        Usage Minimum (Button 1)	19 01 
        Usage Maximum (Button 3)	29 03 
        Logical Minimum (0)	15 00 
        Logical Maximum (1)	25 01 
        Report Size (1)	75 01 
        Report Count (3)	95 03 
        Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit)	81 02 
        Report Count (1)	95 01 
        Report Size (5)	75 05 
        Input (Cnst,Ary,Abs)	81 01 
        Usage Page (Generic Desktop)	05 01 
        Usage (X)	09 30 
        Logical Minimum (0)	15 00 
        Logical Maximum (1913)	26 79 07 
        Physical Minimum (0)	35 00 
        Physical Maximum (1913)	46 79 07 
        Unit Exponent (0)	55 00 
        Unit (SI Lin: )	65 01 
        Report Size (16)	75 10 
        Report Count (1)	95 01 
        Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit)	81 02 
[... 302 lines cut ...]
    Report ID (5)	85 05 
    Usage (Pointer)	09 01 
    Collection (Physical)	A1 00 
        Usage Page (Button)	05 09 
        Usage Minimum (Button 1)	19 01 
        Usage Maximum (Button 3)	29 03 
        Logical Minimum (0)	15 00 
        Logical Maximum (1)	25 01 
        Report Count (3)	95 03 
        Report Size (1)	75 01 
        Input (Data,Var,Abs,NWrp,Lin,Pref,NNul,Bit)	81 02 
        Report Count (1)	95 01 
        Report Size (5)	75 05 
        Input (Cnst,Var,Abs,NWrp,Lin,Pref,NNul,Bit)	81 03 
        Usage Page (Generic Desktop)	05 01 
        Usage (X)	09 30 
        Usage (Y)	09 31 
        Usage (Wheel)	09 38 
        Logical Minimum (-127)	15 81 
        Logical Maximum (127)	25 7F 
        Report Size (8)	75 08 
        Report Count (3)	95 03 
        Input (Data,Var,Rel,NWrp,Lin,Pref,NNul,Bit)	81 06 
    End Collection	C0 
End Collection	C0

Max99, 2015-09-09:
In Windows 10, Is touch input like mouse input system exclusive?

signal11, 2015-09-09:
Maybe. Don't know for sure on Windows 10. Try adjusting the sharing flags.

jherico, 2015-10-20:
I believe this is the same problem reported in https://github.com/thp/psmoveapi/issues/187

Git bisect shows that it broke with this commit: https://github.com/signal11/hidapi/commit/e3ab1210