signal11/hidapi #235: Windows: "unresolved external symbol" if trying to import dll in other solution

jnk0le, 2015-09-20:
``` 
#define HID_API_EXPORT __declspec(dllexport)
```

Shouldn't be used for importing functions from dll library. As I can see it's only working if the hidtest.cpp project exists in the same solution as source of the .dll file comes.

If someone will try to import .dll in another solution it will give  "unresolved external symbol" errors.
Google is full of threads about this kind issue from this library.

This macro should be expanded like this:
```
#ifdef BUILDING_DLL_OR_STH_ELSE_FROM_HIDC
    #define HID_API_EXPORT __declspec(dllexport)
#else
    #define HID_API_EXPORT __declspec(dllimport)
#endif
```

Another thing which can generate same errors is empty
```
#define HID_API_CALL
```
what have to be defined as __cdecl due to people who use eg. __fastcall or any other calling conversion. (__cdecl is set in VS project)

signal11, 2015-09-21:
> Shouldn't be used for importing functions from dll library. As I can see it's only working if the hidtest.cpp project exists in the same solution as source of the .dll file comes.

Not sure what to tell you. These projects have been working in visual studio, mingw, and cygwin for a long time for many, many people, and nobody has complained about this. Being in the same solution doesn't matter. As an experiment: build hidtest.exe, then remove hidapi.vcproj and hidapi.sln (and the .suo, .ncb, and all that stuff); open hidtest.vcproj; and rebuild all. It will build and link fine.

__declspec(dllexport) works for import and for export.

jnk0le, 2015-09-21:
http://www.scriptscoop.com/t/3584644c7224/visual-studio-2010-c-unresolved-external-symbol.html

As I have checked now dllexport works for importing without errors so it shouldn't be a bug, but
> while building the DLL you must use __declspec(dllexport) on the class, and when consuming it there must be __declspec(dllimport). Import libraries have shims to forward free function references to DLLs but those don't work for classes and class members.
Usually macros are used to accomplish the switch between dllexport and dllimport.

The real problem can be calling conversion, depending on default visual settings is not a good idea.
If DLL is built with __cdecl and user is compiling his own project with __stdcall or __fastall, linker will throw "unresolved external symbol" errors.

closed 2015-10-17 by jnk0le