signal11/hidapi #242: Cooperation of hiddev and hidraw

good-king, 2015-10-28:
Hi. I have two processes which communicate with usb hid device. Until now those processes used hiddev interface for this purpose. Now I need to change hiddev to hidraw in one of those processes, and I'm using your library for it. Separately the processes work fine, but when working together the process which is using hidraw is unable to read data from the device, or reads some trash instead of data.  Do you have any ideas why can it happen? OS is linux fedora.

signal11, 2015-10-28:
This may be a lower-level issue than hidapi. I'm not sure it's supported to access a device using hidraw and hiddev at the same time. You can ask this question about using hiddev and hidraw together on the linux-input mailing list. Feel free to put me on CC.

good-king, 2015-10-29:
Thanx for the answer. I asked this question there, but anyway I decided to change everything to hidraw since it doesn't seem to work with hiddev. Are there any features about accessing the same usb hid from different processes using hidraw?

signal11, 2015-10-29:
Should be able to talk from multiple processes with hidraw.

good-king, 2015-10-30:
Just tried. Does hidraw do any synchronization between processes which are using it? Because now when all processes use hidraw, process 1 gets data intended for process 2 and vice versa, everything is messed up.  :S It was not like that with hiddev.

signal11, 2015-10-30:
Then that's a bug in hidraw :(