signal11/hidapi #243: Use libusb calls for hid_write() ?

rytz2k, 2015-11-05:
Greetings:

I'm doing development work on an existing USB device configuration utility that uses both HIDAPI and libusb calls (depending on firmware versions of the device).  libusb calls are "usb_control_msg()" and HIDAPI calls are "hid_write()".

The device is not HID.

Question 1:  would it be possible to migrate the HIDAPI hid_write() calls to usb_control_msg() calls?  I would like to migrate the code into requiring just a single library.

My understanding thus far is that libusb does not support interfacing with HID devices and that HIDAPI is a wrapper for hidraw or libusb, depending on what type of API is required (HID/non-HID).

Thanks.

signal11, 2015-11-07:
I'm not entirely sure what the point of that would be, but you can hack it up any way you see fit. To your last paragraph, look at the code. There are two different implementations on Linux. Pick the one that works for you.

rytz2k, 2015-11-12:
Thanks for the response and info.  I was in the preliminary stages of understanding the various USB libraries and their roles.  Definitely a silly question :)

NicoHood, 2016-03-03:
@signal11 maybe you should close questions like that if they are answered and dont contain a real issue. That would clean the github issue tracker some more.

signal11, 2016-03-03:
I typically let the asker close it.

closed 2016-03-03 by signal11

NicoHood, 2016-03-03:
But this makes the issue tracker very unclear. You should close all invalid issues. And as said, some PRs are really worth and simple to merge. If you could clean up some issues and PRs that'd be nice.