signal11/hidapi #244: Cannot list all devices (Ubuntu 15.04, ASUS)

LuisDa, 2015-11-05:
Hello.

I have downloaded and installed all hidapi libraries, my computer is an Asus laptop with Ubuntu 15.04. When I run the hidtest-hidraw and hidtest-libusb examples (also with root user), only the mouse is listed. I connect an Arduino UNO board, a pendrive, and the program does not recognize them at all.

How can I solve this problem? I would like to be able to develop C or C++ programs that can communicate with my Arduino or with other USB devices.

Thanks.

signal11, 2015-11-06:
Permissions?

Also, many devices are blacklisted from hidraw.

signal11, 2015-11-07:
BTW, arduino and USB thumb drives are not HID devices and won't show up in hidapi.

LuisDa, 2015-11-08:
Hi!

Thanks for your reply :-) .

As you mentioned, I have discovered that Arduino boards are not HID devices, neither external storage disks and pendrives. I have discovered it later (I am kind newbie using HID API).

Regards,

Luisda.

NicoHood, 2015-11-08:
Arduino boards can act as HID device.
See RawHID and (on the dev tree) the feature report keyboard example.
https://github.com/NicoHood/HID

I am currently developing an ambilight via a keyboard feature request. Also see this:
https://github.com/arduino/Arduino/pull/4105

Sharofiddin, 2019-02-22:
Are POS printers HID devices also?