signal11/hidapi #246: linux: libusb method: hid_open() against specific interface number? [code supplied]

rytz2k, 2015-11-12:
The device I'm working with has three interfaces (0-2) and needs to be configured on #2.

I may have missed something with the API but I don't see a way to have hidapi open against a specific interface.  I don't see anything that will let me alter the interface on an opened device, either.

I have this device working properly through both libusb and libhid (when working on interface #2), so I know this is the correct approach.

I've made some simple changes to hid.c that allows me to open with the requirement of a specific interface number, allowing my device to working correctly with hidapi:

```
hid_device * hid_open_find(unsigned short vendor_id, unsigned short product_id, const wchar_t *serial_number, int interface_to_match)
{
	struct hid_device_info *devs, *cur_dev;
	const char *path_to_open = NULL;
	hid_device *handle = NULL;

	devs = hid_enumerate(vendor_id, product_id);
	cur_dev = devs;
	while (cur_dev) {
		if (cur_dev->vendor_id == vendor_id &&
		    cur_dev->product_id == product_id) {

			if (interface_to_match == -1 || interface_to_match == cur_dev->interface_number){
				if (serial_number) {
					if (cur_dev->serial_number &&
					    wcscmp(serial_number, cur_dev->serial_number) == 0) {
						path_to_open = cur_dev->path;
						break;
					}
				}
				else {
					path_to_open = cur_dev->path;
					break;
				}
			}
		}
		cur_dev = cur_dev->next;
	}

	if (path_to_open) {
		/* Open the device */
		handle = hid_open_path(path_to_open);
	}

	hid_free_enumeration(devs);

	return handle;
}

hid_device * hid_open_with_interface(unsigned short vendor_id, unsigned short product_id, const wchar_t *serial_number, int interface){
	return hid_open_find(vendor_id, product_id, serial_number, interface);
}

hid_device * hid_open(unsigned short vendor_id, unsigned short product_id, const wchar_t *serial_number){
	return hid_open_find(vendor_id, product_id, serial_number, -1);
}
```

If there is an existing way to achieve what I have done above, please let me know.

Thanks!

signal11, 2015-11-12:
> I may have missed something with the API but I don't see a way to have hidapi open against a specific interface. I don't see anything that will let me alter the interface on an opened device, either.

"Device" in the HIDAPI parlance really means "interface on a device." That's how all the platform libraries call it, too.

> If there is an existing way to achieve what I have done above, please let me know.

Use hid_enumerate() and find the device/interface you want (using the ->interface field), and then open it with hid_open_path().

rytz2k, 2015-11-12:
Ah, I should have put that combo together as I modified the code.  Thanks for the suggestion!