signal11/hidapi #247: hid_open fails on Windows 7

AndySpider, 2015-11-30:
**hidapi version**: 0.8.0-rc1
**OS**: windows 7 64-bit

I directly modify the hitdtest.cpp, and try to open my device by VID and PID, as following. 
``` C++
handle = hid_open(0x05e0, 0x1200, NULL);       // open my device, the VID and PID are right for sure.
``` 
However, it always failed with a NULL handle. By debugging into the source, I found it fails at `hid.c:228`, which is,
``` C++
	handle = CreateFileA(path,
		desired_access,
		share_mode,
		NULL,
		OPEN_EXISTING,
		FILE_FLAG_OVERLAPPED,/*FILE_ATTRIBUTE_NORMAL,*/
		0);
```
The handle always returns `INVALID_HANDLE_VALUE`.

To eliminate my device problem, I also tried to open a standard keyboard, and it fails as well. By searching around on the web, I found a similar issue [here](https://code.google.com/p/javahidapi/issues/detail?id=46), but it didn't work either.

signal11, 2015-11-30:
HIDAPI won't work with keyboards and mice on Windows.

AndySpider, 2015-12-01:
Why not? I see a lot of cases online in which they need to deal with keyboards, and some of them used HIDAPI.  It should be a big demand, why not support any more, some blockers?

signal11, 2015-12-02:
Security. It's a security problem if a normally-privileged application can keystroke-log the system keyboard. Windows won't allow you to open a keyboard with hidd.dll. If you find a way, let me know.

HIDAPI is for custom-class HID devices, devices which use HID to transfer arbitrary data to a PC.

Thelmos, 2015-12-11:
So a magnetic card reader that emulates a keyboard won't work with HIDAPI?

I have a mouse and a card reader connected to computer and **testgui** don't even list the devices.

Thelmos, 2015-12-11:
Wow!

Changed device mode from *HID Keyboard* to *HID Compatilble* using card reader software and now hidapi works like a charm.

Thanks.

Moonbeem, 2016-10-15:
> hid_open fails on Windows 7
I met the same problem with AndySpider. Could you let me kown the solution? @signal11 @AndySpider

tresf, 2017-02-15:
> Changed device mode from HID Keyboard to HID Compatilble using card reader software and now hidapi works like a charm.

For devices which support this, this is a great option.  Some magnetic wedges also support serial port emulation.

@signal11 this bug should be closed as wont-fix as to clean up the tracker.

Boscop, 2017-11-16:
@Thelmos Which software did you use to change the keyboard's device mode from `HID Keyboard` to `HID Compatible`? (I want to open my Corsair K95 keyboard to get the keystrokes from the additional G keys on it.)

Thelmos, 2017-11-16:
Just the card reader software from the manufacturer