signal11/hidapi #251: On Windows If Trackir software is running first X-Plane crashes using plugin Xsaitekpanels

sparker256, 2016-01-03:
I have had a conflict with Trackir software and my Xsaitekpanels on Windows and finally found the issue and have a patch. I had resolved the issue but did not realize that I was using release hidapi-0.7.0 and as soon as I tried to use hidapi-0.8.0-rc1 the crash was back. I went through commits from the release of hidapi-0.7.0 until at git checkout e3ab121 the crash was back. Here is my diff and sorry it took me so long to find it.

Bill
hidapi-0.8.0-rc1_windows_hid.diff.zip

signal11, 2016-09-20:
Hi Bill,

This stuff is all different now. Basically every device is opened with sharing since b5b2e1779b6cd2edda3066bbbf0921a2d6b1c3c0 . Do you still see this issue with the latest from git (git clone ...) ?

sparker256, 2016-09-20:
It looks like I am not using the latest git clone version and will retest and get back to you.

sparker256, 2016-09-21:
Done testing using the latest git clone and my issue has been resolved so I am closing this issue.

closed 2016-09-21 by sparker256