signal11/hidapi #255: hid_write REPORT_ID byte on linux/windows/mac

jonasschnelli, 2016-01-29:
I'm currently finalizing deployment of a cross platform application.
But what I can't follow is,...

Why do i need to send the REPORT_ID on mac/windows:
(pseudecode):
`hid_write(handle, 0x00 + buffer, size(buffer)+1)`

While on Linux it need to be
`hid_write(handle, buffer, size(buffer))`

Can I somehow detect if hid_write expect a REPORT_ID?

signal11, 2016-02-01:
It should be the same. Are you using the latest from git?

jonasschnelli, 2016-02-23:
I'm using 0.8.0-rc1.
Maybe it was fixed with https://github.com/signal11/hidapi/commit/54eb31dc16dcc67d0b689ed947bc53a038608c0e which is not in 0.8.0-rc1.

What about doing a 0.8.0 release (current master)?

jonasschnelli, 2016-02-25:
It seems to be a difference between `hid-libusb` and `hid-hidraw` (at least on linux).

jonasschnelli, 2016-02-25:
https://github.com/signal11/hidapi/commit/3a66d4e513ed1b1ce82b7e6fcfa30ff05598b696 this commit is un Ubuntus version but not in the 0.8.0rc1 tag. Maybe spun of a rc2?

swt2c, 2016-03-24:
Indeed.  Ubuntu < 16.04 has a git snapshot at rev 3a66d4e.  Ubuntu 16.04+ has a git snapshot at rev d17db57.

nvchernov, 2018-06-01:
So much thanks for this question!!! I found solution for me with your approach.
Function works great, but it behaves really unexpecting!