signal11/hidapi #256: Mac: hid_enumerate lists all devices twice

garrettr, 2016-02-03:
Mac 10.10.5
HIDAPI Version: d17db57b9d4354752e0af42f5f33007a42ef2906 (current master)

## STR

I adapted the example code to simply call hid_enumerate and list some details about each of the available devices, `example.c`:

```c
// adapted from https://github.com/signal11/hidapi/blob/master/README.txt

#include <stdio.h>
#include <stdlib.h>
#include "hidapi.h"

int main(int argc, char* argv[])
{
  int res;

  // Initialize the hidapi library
  res = hid_init();

  // Enumerate and print all the HID devices on the system
  struct hid_device_info *devs, *cur_dev;

  devs = hid_enumerate(0x0, 0x0);
  cur_dev = devs;
  while (cur_dev) {
    printf("Device Found\n  type: %04hx %04hx\n  path: %s\n  serial_number: %ls",
           cur_dev->vendor_id, cur_dev->product_id, cur_dev->path, cur_dev->serial_number);
    printf("\n");
    printf("  Manufacturer: %ls\n", cur_dev->manufacturer_string);
    printf("  Product:      %ls\n", cur_dev->product_string);
    printf("\n");
    cur_dev = cur_dev->next;
  }
  hid_free_enumeration(devs);

  return 0;
}
```

I put this in `hidapi/mac` and built it with `clang -g -D_THREAD_SAFE -pthread -framework IOKit -framework CoreFoundation hid.c example.c -I../hidapi -o example`

## Observed output

```
$ ./example                                                                                                         
Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad

Device Found
  type: 1050 0407
  path: USB_1050_0407_14200000
  serial_number: 
  Manufacturer: Yubico
  Product:      Yubikey 4 OTP+U2F+CCID

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad

Device Found
  type: 24f0 0140
  path: USB_24f0_0140_14140000
  serial_number: 
  Manufacturer: Metadot - Das Keyboard
  Product:      Das Keyboard

Device Found
  type: 24f0 0140
  path: USB_24f0_0140_14140000
  serial_number: 
  Manufacturer: Metadot - Das Keyboard
  Product:      Das Keyboard

Device Found
  type: 1050 0407
  path: USB_1050_0407_14200000
  serial_number: 
  Manufacturer: Yubico
  Product:      Yubikey 4 OTP+U2F+CCID

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad

Device Found
  type: 1a7c 0191
  path: USB_1a7c_0191_14120000
  serial_number: 
  Manufacturer: HOLTEK
  Product:      Evoluent VerticalMouse 4
```

As you can see, each device is listed twice.

## Expected output

I would expect each attached device to only be listed once in the linked list returned by `hid_enumerate`.

I added a debugging `printf` to `hid.c` and confirmed that `num_devices` from `num_devices = CFSetGetCount(device_set);` is 8; I would expect it to be 4, since there are 4 HID devices currently attached to my system.

garrettr, 2016-02-03:
Actually, I see now that not every device is listed twice. In my case:

1. The Apple Internal Keyboard / Trackpad is listed 3x
2. The Das Keyboard is listed 2x
3. The Yubikey is listed 2x
4. The Evoluent VerticalMouse 4 is listed 1x

The order changes every time I run `./example` but the profile of repeated devices is consistent each time.

signal11, 2016-02-03:
Very strange. Are you using the latest from git?

Are you able to debug into it?

garrettr, 2016-02-03:
I am using the latest commit fro master. Is there a different branch I should be using? I can take a look at debugging it now.

garrettr, 2016-02-03:
I added some more debugging output and it appears that each physical device gets multiple IOHidDevices created for it, approximately one for each usage (although the Apple Internal Keyboard still has a duplicate entry for `Usage Page:   0001, Usage:        0002`:

```
$ ./example                                                                                                         
hid_enumerate: num_devices=6
Device Found
  type: 1050 0407
  path: USB_1050_0407_14100000
  serial_number: 
  Manufacturer: Yubico
  Product:      Yubikey 4 OTP+U2F+CCID
  Usage Page:   f1d0
  Usage:        0001

Device Found
  type: 1050 0407
  path: USB_1050_0407_14100000
  serial_number: 
  Manufacturer: Yubico
  Product:      Yubikey 4 OTP+U2F+CCID
  Usage Page:   0001
  Usage:        0006

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   0001
  Usage:        0002

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   0001
  Usage:        0002

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   ff00
  Usage:        0001

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   0001
  Usage:        0006
```

This sort of makes sense to me, although it is not what I initially expected (although I am new to HID programming in general, so perhaps that is the reason why it was surprising).

garrettr, 2016-02-03:
One thing that still appears to be a bug: none of the devices have a serial number! Should I file a separate issue for that? Investigating now.

garrettr, 2016-02-03:
Examined in lldb, looks like IOHIDDeviceGetProperty returns NULL for the serial number. This seems like a bug, but not in hidapi. I'm not sure if there's anything we can do about it :disappointed:

```
$ lldb example
(lldb) target create "example"
Current executable set to 'example' (x86_64).
(lldb) b get_serial_number 
Breakpoint 1: where = example`get_serial_number + 27 at hid.c:309, address = 0x0000000100001f1b
(lldb) r
Process 14092 launched: '/Users/garrett/code/hidapi/mac/example' (x86_64)
hid_enumerate: num_devices=6
Process 14092 stopped
* thread #1: tid = 0x19922, 0x0000000100001f1b example`get_serial_number(device=0x0000000100306f90, buf=L"", len=256) + 27 at hid.c:309, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
    frame #0: 0x0000000100001f1b example`get_serial_number(device=0x0000000100306f90, buf=L"", len=256) + 27 at hid.c:309
   306 	
   307 	static int get_serial_number(IOHIDDeviceRef device, wchar_t *buf, size_t len)
   308 	{
-> 309 		return get_string_property(device, CFSTR(kIOHIDSerialNumberKey), buf, len);
   310 	}
   311 	
   312 	static int get_manufacturer_string(IOHIDDeviceRef device, wchar_t *buf, size_t len)
(lldb) s
Process 14092 stopped
* thread #1: tid = 0x19922, 0x000000010000351b example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 27 at hid.c:229, queue = 'com.apple.main-thread', stop reason = step in
    frame #0: 0x000000010000351b example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 27 at hid.c:229
   226 	{
   227 		CFStringRef str;
   228 	
-> 229 		if (!len)
   230 			return 0;
   231 	
   232 		str = IOHIDDeviceGetProperty(device, prop);
(lldb) b 232
Breakpoint 2: where = example`get_string_property + 53 at hid.c:232, address = 0x0000000100003535
(lldb) c
Process 14092 resuming
Process 14092 stopped
* thread #1: tid = 0x19922, 0x0000000100003535 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 53 at hid.c:232, queue = 'com.apple.main-thread', stop reason = breakpoint 2.1
    frame #0: 0x0000000100003535 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 53 at hid.c:232
   229 		if (!len)
   230 			return 0;
   231 	
-> 232 		str = IOHIDDeviceGetProperty(device, prop);
   233 	
   234 		buf[0] = 0;
   235 	
(lldb) p str
(CFStringRef) $0 = 0x00007fff5fbfee80
(lldb) n
Process 14092 stopped
* thread #1: tid = 0x19922, 0x0000000100003546 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 70 at hid.c:234, queue = 'com.apple.main-thread', stop reason = step over
    frame #0: 0x0000000100003546 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 70 at hid.c:234
   231 	
   232 		str = IOHIDDeviceGetProperty(device, prop);
   233 	
-> 234 		buf[0] = 0;
   235 	
   236 		if (str) {
   237 			CFIndex str_len = CFStringGetLength(str);
(lldb) p str
(CFStringRef) $1 = 0x0000000000000000
(lldb) n
Process 14092 stopped
* thread #1: tid = 0x19922, 0x0000000100003550 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 80 at hid.c:236, queue = 'com.apple.main-thread', stop reason = step over
    frame #0: 0x0000000100003550 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 80 at hid.c:236
   233 	
   234 		buf[0] = 0;
   235 	
-> 236 		if (str) {
   237 			CFIndex str_len = CFStringGetLength(str);
   238 			CFRange range;
   239 			CFIndex used_buf_len;
(lldb) n
Process 14092 stopped
* thread #1: tid = 0x19922, 0x0000000100003640 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 320 at hid.c:263, queue = 'com.apple.main-thread', stop reason = step over
    frame #0: 0x0000000100003640 example`get_string_property(device=0x0000000100306f90, prop=@"SerialNumber", buf=L"", len=256) + 320 at hid.c:263
   260 			return 0;
   261 		}
   262 		else
-> 263 			return -1;
   264 	
   265 	}
   266 	
(lldb) 
```

garrettr, 2016-02-04:
@signal11 Should I file a separate issue for the missing serial number on Mac OS X, and close this one? I think my original concern about devices being listed multiple times was invalid, that's just how devices with multiple usages are presented AFAICT.

closed 2016-02-09 by garrettr