signal11/hidapi #257: Serial number unavailable on Mac

garrettr, 2016-02-09:
Separating this into its own issue from where it was originally reported in #256.

OS: Mac 10.10.5
HIDAPI version: d17db57b9d4354752e0af42f5f33007a42ef2906 (current master)

## STR

I'm running the following example program:

```c
// adapted from https://github.com/signal11/hidapi/blob/master/README.txt

#include <stdio.h>
#include <stdlib.h>
#include "hidapi.h"

int main(int argc, char* argv[])
{
  int res;

  // Initialize the hidapi library
  res = hid_init();

  // Enumerate and print the HID devices on the system
  struct hid_device_info *devs, *cur_dev;

  devs = hid_enumerate(0x0, 0x0);
  cur_dev = devs;
  while (cur_dev) {
    printf("Device Found\n  type: %04hx %04hx\n  path: %s\n  serial_number: %ls",
           cur_dev->vendor_id, cur_dev->product_id, cur_dev->path, cur_dev->serial_number);
    printf("\n");
    printf("  Manufacturer: %ls\n", cur_dev->manufacturer_string);
    printf("  Product:      %ls\n", cur_dev->product_string);
    printf("  Usage Page:   %04hx\n", cur_dev->usage_page);
    printf("  Usage:        %04hx\n", cur_dev->usage);
    printf("\n");
    cur_dev = cur_dev->next;
  }
  hid_free_enumeration(devs);

  return 0;
}
```

I've put it in the `mac/` subdirectory and built it with `clang -g -D_THREAD_SAFE -pthread -framework IOKit -framework CoreFoundation hid.c example.c -I../hidapi -o example`.

## Output

This is just running on my Macbook Pro laptop with a Yubico Yubikey external device attached.

```
Device Found
  type: 1050 0407
  path: USB_1050_0407_14200000
  serial_number: 
  Manufacturer: Yubico
  Product:      Yubikey 4 OTP+U2F+CCID
  Usage Page:   f1d0
  Usage:        0001

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   0001
  Usage:        0002

Device Found
  type: 1050 0407
  path: USB_1050_0407_14200000
  serial_number: 
  Manufacturer: Yubico
  Product:      Yubikey 4 OTP+U2F+CCID
  Usage Page:   0001
  Usage:        0006

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   0001
  Usage:        0002

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   ff00
  Usage:        0001

Device Found
  type: 05ac 0259
  path: USB_05ac_0259_14500000
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Usage Page:   0001
  Usage:        0006

```

## Expected Output

The device's serial number should be available via `cur_dev->serial_number`.

## Observed Output

No device has a serial number available - `cur_dev->serial_number` is an empty string.

garrettr, 2016-02-09:
I ran `./example` with a debugger and reported the output in https://github.com/signal11/hidapi/issues/256#issuecomment-179392839. It shows that `IOHIDDeviceGetProperty` returns a null pointer when queried with `prop=@"SerialNumber"`, so I'm not sure if this is a problem with the native API or if the native API has changed.

garrettr, 2016-02-09:
The same issue appears to have been reported here (in 2013): http://www.lists.apple.com/archives/usb/2013/Apr/msg00018.html

matthewbauer, 2017-03-08:
Can you confirm that the serial number shows up in "System Information" for the device? It's possible that some devices don't show the serial number.