signal11/hidapi #262: Feature Report not working under Windows 7

gregwing, 2016-03-08:
I am running the test app as supplied with the library zip, using win7 / MSVC2013.I am able to scan the USB and open my joystick device, device created with GenericHID. It it is able to read the raw data coming from the USB controls but the feature_report does not work. all I ever get back from this function is "Incorrect function"

I have modified the code to request a range of report numbers but get the same response for all.

	// Read a Feature Report from the device
	for (int rep = 0; rep < 0x91; rep++)
	{
		buf[0] = rep;
		res = hid_get_feature_report(handle, buf, sizeof(buf));
		if (res < 0)
		{
			printf("Unable to get a feature report %d.\n", rep);
			printf("%ls\n", hid_error(handle));
		}
		else
		{
			// Print out the returned buffer.
			printf("Feature Report %d\n   ", rep);
			for (i = 0; i < res; i++)
				printf("%02hhx ", buf[i]);
			printf("\n");
		}
	}

It does not say anywhere what the report number means, is it the interface number, the code page or what?

Does some thing need to be set up / enabled before the request will work? I assume not as there is not anything in this demo code?

Cheers Greg

signal11, 2016-03-08:
> I am running the test app as supplied with the library zip,

Run the latest from git.

> It does not say anywhere what the report number means, is it the interface number, the code page or what?

It's the report number. See the HID specification. It's used for devices with multiple reports.

> Does some thing need to be set up / enabled before the request will work?

The device needs to support feature reports. Most devices just don't have any. I wouldn't expect a commercial joystick to have any.

gregwing, 2016-03-08:
> Run the latest from git.
OK, tried that, no change

> The device needs to support feature reports. Most devices just don't have any. I wouldn't expect a > commercial joystick to have any.
It was my impression that the reports contain the usage information. The usage information tells you how the data read from the device relates to the buttons and sliders on the device (as per the HID spec Appendix E (E.10)). Is that another kind of report?

signal11, 2016-09-20:
Reports contain the values for each usage. The report descriptor describes the formats of the reports.