signal11/hidapi #264: Multi threaded use of HidAPI

brebisson, 2016-03-14:
Hello,

I have an issue in a multi-threaded program using HidAPI.

I have a read thread which is waiting on USB coming in, and a monitoring thread which, among other things receives the order to 'terminate USB connections'.

At this point, I need to stop the communication. I do this by doing a hid_close.

This correctly stops the underlying ReadFile calls, BUT the hid_read call then proceed to write in the hid_device memory area (dev->read_pending = FALSE;).
This of course is bad because hid_close has just freed that area of memory...

So, my question is: how do I tell a hid_read to 'stop' without doing a hid_close (and freeing the structure)?

It seems that the best architectural solution would be a change to the hid_device structure to have some 'use counter' that would let multiple hid calls use the structure at the same time and handle correct freeing of the structure when needed. Something around the lines of:
EnterCriticalSection; Inc_Counter; LeaveCriticalSection; at the top of each hid_call that uses hid_device.
and EnterCriticalSection; dec_Counter; if (counter==0) free(structure); LeaveCriticalSection;
type thing.

Any ideas?

Cyrille

signal11, 2016-03-14:
> It seems that the best architectural solution would be a change to the hid_device structure to have some 'use counter' 

It would be better of course to use real mutexes.

androjes, 2017-01-01:
There is a way to open multi devices independently ?