signal11/hidapi #268: Opening a device with kIOHIDOptionsTypeSeizeDevice should be optional on Mac

garrettr, 2016-03-29:
I don't really understand the reasoning behind using `kIOHIDOptionsTypeSeizeDevice` to gain exclusive access to devices on Mac, especially since:

1. That's [no longer being done on Windows](https://github.com/signal11/hidapi/commit/b5b2e1779b6cd2edda3066bbbf0921a2d6b1c3c0)
2. It was never done on Linux (no `flock` in the Linux implementation of [`hid_open_path`](https://github.com/signal11/hidapi/blob/master/linux/hid.c#L627)

...so this behavior is not consistent cross-platform.

Furthermore, it creates issues with certain use cases. For example, I am trying to implement U2F (a two-factor authentication protocol that uses hardware security tokens which are currently commonly implemented as HID devices, e.g. Yubikey) in Firefox. Chrome already has U2F implemented, and [does not use exclusive access on Mac](https://code.google.com/p/chromium/codesearch#chromium/src/device/hid/hid_service_mac.cc&sq=package:chromium&q=iohiddeviceopen&l=173).

Since it is common for users to use more than one browser, sometimes simultaneously, this could create a potential problem where Firefox's U2F implementation, if it uses HIDAPI and therefore `SeizeDevice`, might prevent Chrome's U2F implementation from connecting to the underlying device and thus render Chrome unable to authenticate users via U2F.

This is certainly an edge case, but I can almost see the Bugzilla bug now. It seems like a potentially frustrating case for users that should be avoided if only because I can't understand why we need to seize the device on Mac in the first place.

signal11, 2016-09-20:
Seizing the device solves the problem of having two programs trying to talk to the device at once and not knowing about one another. There's only one device. How can two programs set states on that device without knowing about one another in a coherent way?

That said, this had to be removed on Windows, because an increasing number of devices (and possibly all devices on windows 10) could not be opened this way. It might be time to revisit this on Mac.

signal11, 2016-09-20:
@mrpippy what do you think about this?

amadsen, 2016-11-12:
U2F devices are supposed to support "transactions" with a nonce so that they can tell which host program (U2F client) is communicating with them. During a transaction, the device will reject communication from another host program, but otherwise the device is supposed to be available.

At least 2 different of u2f libraries are using hidapi at this point, including Yubico's [u2f-host](https://github.com/Yubico/libu2f-host/tree/master/u2f-host) which is used by an existing Firefox u2f extension and [u2f-client](https://github.com/ashtuchkin/u2f-client) through [node-hid](https://github.com/node-hid/node-hid). This would help address one issue they encounter.

(The recent fix to the OSX path seems to have addressed a different issue with the multifunction U2F capable devices. Thanks!)

herrjemand, 2017-07-20:
@amadsen did you manage to run u2f-client on the mac? I had no success so far... *(