signal11/hidapi #270: Makefile-manual compile error

sat-turn, 2016-04-03:
hi, im a bit of a newbie, i have followed the install manual given however when i manually try to build it, it gives this error. any help would be greatly appreciated!
[image]

signal11, 2016-04-03:
Never seen that one before. I don't know what your setup is so I can't be sure.

Did you try adding -fPIC as the error message suggested?

sat-turn, 2016-04-03:
thanks for replaying, my setup is a vm running ubuntu 15.10,  i have tried adding -fPIC no luck 
[image]

signal11, 2016-04-03:
Put the -fPIC in the makefile. It's a compiler flag. See the man page for gcc.

If you're going to use the shared library, why not run the ./configure and install it the normal way if you're on Linux?

sat-turn, 2016-04-03:
oh i see, im quite new to this so i might be making simple mistakes. Tried installing the normal way by 
./configure && make && sudo make install
it seems to have succeeded, here is the output. http://paste.ofcode.org/34FNGi63nsbe3UTVgBwx55b
however when i run hid.c, initially its missing hidapi.h after i paste it in the folder from the hidapi folder, the output is this. 
[image]

SlySven, 2016-04-03:
**sat-turn** wrote:
>...however when i run hid.c...

Um, that output is from you attempting to *compile* hic.c - a source file , manually rather than to **run** it - and it is failing because you ARE running that compilation stage manually.  If you had done that step using the *make* system it would have run the equivalent of that command with *a lot* more arguments including `-I` ones that tell the *compiler* where to find the header files (the `#include <xxx.h>` and possibly some `#include "xxx.h"` ones as well) items in the source code files AND `-l` and `-L` ones that tell the step after that, the *linker* the names of libraries and their locations where the functions that are not in a particular source file can be found.

Those error messages are from the linker which is trying to link the `hid.o` object file that the compiler makes from the `hid.c` source file to make an `hid` executable and is moaning about all the functions that it cannot find.

tl;dr;

Re-read the documentation on how to build the project - (I think) you are doing things wrong...!