signal11/hidapi #271: USB Keyboard/Raw device - /dev/input/eventX vanishes when hid_open() is called.

woehrl, 2016-04-05:
Hope you could give me a hint. We use a combinated USB HID device for keyboard usage and other reports with Qt 5.5 on a Raspberry Pi 3 (using Raspbian).
Problem is - The keyboard device (dev/input/event0) gets disconnected as soon as I open the hid_device with:
hid_device *deviceHandle = hid_open(RPIBASE_VENDORID, RPIBASE_PRODUCTID, NULL); 

When the app is started the device is connected and Qt grabs it:
qt.qpa.input: Opening keyboard at "/dev/input/event0"

But as soon as we open (hid_open) or send reports (hid_write) with hidapi to the same device, the keyboard just gets disconnected (/dev/input/event0 really vanishes) and Qt disconnects, too:
qt.qpa.input: Removing keyboard at "/dev/input/event0"
qt.qpa.input: Unload current keymap and restore built-in
evdevkeyboard: Failed to query led states
qt.qpa.input: switchLed 0 false
qt.qpa.input: switchLed 1 false
qt.qpa.input: switchLed 2 false

I would expect a complaint by hidapi or the kernel over concurrent access or so but nothing is found in dmeg/message when that happens...

Anyone a clue?

woehrl, 2016-04-06:
After having a conversation with Alan and Scott I realized, that I used the hidapi-libusb implementation. After switching and linking to the hidapi-hidraw implementation it runs fine:
`LIBS += -lhidapi-hidraw`
Thanks!

closed 2016-04-06 by woehrl

NicoHood, 2016-04-07:
@woehrl Just a quick question:
I am currently using `libhidapi` from the ubuntu repositories (Installed `libhidapi*`).
However my program doees not find `#include <hidapi.h>`. I used `-lhidapi-hidraw` or `-lhidapi` as gcc flag. `-lusb` for using libusb works though.

Do you know what I am doing wrong? I (preferred) do not want a local copy of hidapi, i'd rather use the built in package.

woehrl, 2016-04-07:
Did you install package `libhidapi-dev`? And for the right platform? (on ubuntu there might be `libhidapi-dev` and `libhidapi-dev:i386`)...
Then you should have a "hidraw.h" header file on your system - normally in /usr/include/linux/ (you may find it with `locate hidraw.h`. To compile you need a header file - for distribution, the library is sufficient.
You might need to add this include path then to your Makefile.

NicoHood, 2016-04-07:
```
libhidapi-dev is already the newest version.
libhidapi-hidraw0 is already the newest version.
libhidapi-hidraw0-dbg is already the newest version.
libhidapi-libusb0 is already the newest version.
libhidapi-libusb0-dbg is already the newest version.
```

It is inside `/usr/include/linux/hidraw.h`
I though I can just pass `lhidapi` to the compiler like with libusb. Isn't there such a shortcut?

woehrl, 2016-04-08:
I never tried on ubuntu, but that is standard linking procedure. Try to add:
```
INCLUDEPATH += /usr/include/linux/
LIBS += -L"/usr/lib/x86_64-linux-gnu/" -lhidapi-hidraw
```
There might also be a way to use pkgconfig, but you might have to google on this...
Good luck, best regards!

signal11, 2016-04-08:
or better, use pkg-config

NicoHood, 2016-04-08:
Maybe with more details? I could not find any information about this online.

signal11, 2016-04-08:
> I could not find any information about this online.

Google yields:
  https://en.wikipedia.org/wiki/Pkg-config
  https://people.freedesktop.org/~dbn/pkg-config-guide.html

woehrl, 2016-04-09:
That is exactly what pkgconfig tells you, right:
`$ pkg-config --libs --cflags hidapi-hidraw`
-> `-I/usr/include/hidapi -lhidapi-hidraw`
Use it in your compile statement:
`$ gcc -o test test.c $(pkg-config --libs --cflags hidapi-hidraw)`
(from wikipedia)