signal11/hidapi #273: I guess testgui misses Fox include path and link library lines?

procedural, 2016-04-12:
Adding them manually to Makefile like so compiles without errors for me on Ubuntu 16.04:

```
DEFAULT_INCLUDES = -I. -I$(top_builddir) -I/usr/include/fox-1.6/
```

```
CXXLINK = $(LIBTOOL) $(AM_V_lt) --tag=CXX $(AM_LIBTOOLFLAGS) \
	$(LIBTOOLFLAGS) --mode=link $(CXXLD) $(AM_CXXFLAGS) \
	$(CXXFLAGS) $(AM_LDFLAGS) $(LDFLAGS) -L/usr/lib/x86_64-linux-gnu/ -lFOX-1.6 -o $@
```

signal11, 2016-04-12:
Does pkg-config or fox-config no longer work?

procedural, 2016-04-12:
@signal11  it does work for me:
```bash
$ pkg-config --cflags --libs fox
-I/usr/include/fox-1.6 -lFOX-1.6
```
But I don't know whether you use it in Makefile or not, I'm not familiar with it.