signal11/hidapi #275: hid_open_path return same handle for two different devices

RomainRivoire, 2016-05-04:
Hi, 
I have a Java application that communicates with hid usb devices using JNI calling hid.c functions,
on a windows 10 pro, the behavior is correct. But on a windows 10 tablet with home edition windows version, it loops on the same device cause the handle returned by "hid_open_path" is the same
although I'm passing a different path argument (I debugged the JNI side redirecting the output to a txt file)

Here is the JNI code 

           hid_device *handle;
		handle = 0;
        struct hid_device_info *devs, *cur_dev;
        
      	devs = hid_enumerate(0x0, 0x0);
        cur_dev = devs;
        const char* serial = (*env)->GetStringUTFChars(env,javaserial,0);
        
        /* Iterating over all the connected devices and finding the one we want */
        while (cur_dev) {
      		
            if(cur_dev->vendor_id == vid && cur_dev->product_id == pid) {

                /* If we have an interface number, check it to make sure it matches */
                if(device >= 0 && cur_dev->interface_number != device) {
                    cur_dev = cur_dev->next;
                    continue;
                }
                
                /* Same goes with the serial */
                if(cur_dev->serial_number && javaserial != NULL) {

					printf("vendor_id=%d\n", cur_dev->vendor_id);
					printf("product_id=%d\n", cur_dev->product_id);
					printf("product_id=%d\n", cur_dev->serial_number);
                    
                    jstring curjavaSerial= (*env)->NewString(env,cur_dev->serial_number,wcslen(cur_dev->serial_number));
                    const char* curSerial=(*env)->GetStringUTFChars(env,javaserial,0);
                    
                    if(strcmp(serial,curSerial) != 0) {
                        cur_dev = cur_dev->next;
                        continue;
                    }
                }

				
                handle = hid_open_path(cur_dev->path);  
                hid_set_nonblocking(handle, 1);

                
                if(!handle) {
                    printf("HID open of path failed\n");
                    hid_free_enumeration(devs);
                    return -1;
                }

				printf("Handle connect device : %d \n",handle);
                
                hid_free_enumeration(devs);
                return (int)handle;
            }
    
            cur_dev = cur_dev->next;
        }
    
        hid_free_enumeration(devs);        
        printf("Could not find HID device\n");
        return -1;

amullins83, 2016-05-04:
When does the value of `javaserial` ever change inside the loop? I see this before the loop:

```C
const char *serial = (*env)->GetStringUTFChars(env, javaserial, 0);
```

Not being a JNI developer, I can only assume this extracts a string from the Java environment, but then we access the same string here inside the loop:

```C
const char *curSerial = (*env)->GetStringUTFChars(env, javaserial, 0);
```

And then we compare the two (presumably identical) strings:

```C
if (strcmp(serial, curSerial) != 0) { ... }
```

When would these two strings ever not be equal? What am I missing here?

signal11, 2016-05-04:
> I debugged the JNI side redirecting the output to a txt file

What did you learn from this debugging? Have you stepped through the hid.c part? Have you tried just a simple C program?

RomainRivoire, 2016-05-04:
The javaserial is a parameter passed from Java to the JNI side.

From the debugging, I have learnt that even if, randomly, the handles are different, it aims to the same physical device anyways.

The Java side is simple, I call in a loop  : 
for each device ! connectDevice->sendData->disconnectDevice

On the JNI side, I see sometimes the same handle for both devices, sometimes an unique handle for each one (which is the correct behavior), but in both cases, the target is the same physical device.

I can't figure out what's going in the hid_open function, I have tried to change the code according to the #231 issue but it's not changing anything here.

This JNI side is provided by Texas Instruments, Java_Hid_Demo [http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/MSP430_USB_Developers_Package/latest/index_FDS.html](url)

And the DLL 

[com_ti_msp430_usb_hiddemo_management_HidCommunicationManager.txt](https://github.com/signal11/hidapi/files/248916/com_ti_msp430_usb_hiddemo_management_HidCommunicationManager.txt)

signal11, 2016-05-04:
If you're not using the latest, start with that. See what happens in a simple C program in hid.c.

RomainRivoire, 2016-05-04:
I am using the latest, the code is working fine on windows 7.
I'll try to write an even more C program

signal11, 2016-05-04:
If you were trying to incorporate the changes from #231, you're not using the latest, because those are in the latest.

RomainRivoire, 2016-05-10:
I wasn't using the last one but now I do, my debugging shows me that hid_enumerate gives me the same path for both devices, so it's logical.
This only happens on Windows 10

signal11, 2016-05-10:
Is it a bug in Windows? Is Windows giving you the same path for the two devices? Have you stepped into hid_enumerate() with a debugger?

RomainRivoire, 2016-05-10:
When I connected one after one my devices, the path is different, but when I plug them both the path is the same with the last version of hidapi. 
Serials are different but path is exactly the same.

signal11, 2016-05-10:
My last response still applies. Have you stepped into hid_enumerate() with a debugger and looked at what Windows is giving you for paths?

RomainRivoire, 2016-05-12:
Windows is giving the right paths, 
it is actually an error in the DLL provided by Texas Instruments : 

`   jstring curjavaSerial= (*env)->NewString(env,cur_dev->serial_number,wcslen(cur_dev->serial_number));
                const char* curSerial=(*env)->GetStringUTFChars(env,javaserial,0);` 

_javaserial_ should be _curjavaSerial_

closed 2016-05-12 by RomainRivoire

signal11, 2016-05-12:
Ah. Glad you got to the bottom of it.

amullins83, 2016-05-12:
@RomainRivoire Have you contacted anyone at TI about this yet? I know a few people there who might be interested in fixing this bug.

RomainRivoire, 2016-05-13:
@amullins83 I did, I opened a topic in the msp430 forum about that problem and suggested this fix as an answer.