signal11/hidapi #276: (Windows10) CreateFileA fails for pen device

p-i-, 2016-05-14:
I've documented this fully on SO http://stackoverflow.com/questions/37213634/createfilea-fails-to-open-hid-device-in-windows

I'm linking it here as it contains research that may help someone else. Also, maybe someone can see a solution?

My pen device actually reports 2 separate devices when I enumerate: UsagePage=1, Usage=2 and UsagePage=13, Usage=2.

hidapi currently doesn't let you open one of these, it will simply open the first vid/pid match.  So it might be an idea to allow a couple of extra optional parameters to give a finer grain control.

If I attempt to open the 1/2 path I get ERROR_ACCESS_DENIED
If I attempt to open the 13/2 path I get ERROR_SHARING_VIOLATION

1/2 is {Generic Desktop Page}/{Mouse} and I have learned that Windows forbids read/write access on Mouse/Keyboard as a security measure.
However, I can see no reason why the 13/2 path fails. It seems to be suggesting some of the process has opened this path with conflicting access/sharing permissions.  I have read that Windows opens all HID devices upon detection, so maybe if there was some way to force it to close this device before opening it...

π

signal11, 2016-05-16:
Each "device" that you see shows up as a separate device to HIDAPI. This is how the platform APIs work. Each USB Interface that reports itself as HID will show up as a "device" in HIDAPI. You can differentiate between them by using the data returned from hid_enumerate().

signal11, 2016-09-20:
Closing from no response.

closed 2016-09-20 by signal11