signal11/hidapi #281: When the USB device is removed, how to know the device has been removed?

tyzthedoors, 2016-06-22:
Hi!
I want to know, When the USB device is removed on mac ox, how to know the device has been removed?
Thank you！

signal11, 2016-06-22:
hid_read() fails when the device is removed.

tyzthedoors, 2016-06-23:
Thank you for your answer！
But I want to know:
The mac ox can not take the initiative to know the device is removed?
For example, I want to monitor the status of device, I have to cycle call hid_read ()? 
No other way?
For example, Using block？

signal11, 2016-06-23:
You're polling hid_read() anyway to see if there's data; why not just use the result? If there's data, process it (because you need to anyway), and if it's -1, then the device is removed.

If you want something different, you can modify the library. There are removal callbacks internal to HIDAPI on OSX. Feel free to hook up to those, but it won't be portable.

tyzthedoors, 2016-06-24:
I looked at Apple's official documentation,I found a way -->#include <IOKit/hid/IOHIDManager.h>
Is this the method is called when the USB remove?

/*! @function   IOHIDManagerRegisterDeviceRemovalCallback
    @abstract   Registers a callback to be used when any enumerated device is 
                removed.
    @discussion In most cases this occurs when a device is unplugged.
    @param      manager Reference to an IOHIDManagerRef.
    @param      callback Pointer to a callback method of type 
                IOHIDDeviceCallback.
    @param      context Pointer to data to be passed to the callback.
*/
CF_EXPORT
void IOHIDManagerRegisterDeviceRemovalCallback(
                                IOHIDManagerRef                 manager,
                                IOHIDDeviceCallback             callback,
                                void *                          context)
AVAILABLE_MAC_OS_X_VERSION_10_5_AND_LATER;

If this is the case, can you use this method to package a api monitor usb removal？

tyzthedoors, 2016-06-24:
include --> IOKit/hid/IOHIDManager.h

tyzthedoors, 2016-06-24:
If I polling hid_read(),It may result in unnecessary waste,And get the result will be have some error.

amullins83, 2016-06-24:
> If I polling hid_read(),It may result in unnecessary waste

How are you using HIDAPI now? Are you not polling `hid_read()` to get data from a device?

> And get the result will be have some error.

Right. Errors are always possible during device communication. Being disconnected is one of them.

I think you've found the correct function library for hooking Mac OSX events (I can't really verify as I no longer own a Mac). If your application only targets OSX, you should feel free to use those. HIDAPI is multi-platform, and since messaging is very different across platforms, there's very little chance any such hooks can be added to the API itself.

tyzthedoors, 2016-06-25:
static void Handle_DeviceMatchingCallback(void *inContext,IOReturn inResult,void *inSender,IOHIDDeviceRef inIOHIDDeviceRef) {
    printf("%p-->Device Into",(void *)inIOHIDDeviceRef);
}
static void Handle_DeviceRemovalCallback(void *inContext,IOReturn inResult,void *inSender,IOHIDDeviceRef inIOHIDDeviceRef) {
    printf("%p-->Device Remove",(void *)inIOHIDDeviceRef);
}

if (hid_mgr)
{
......(When the function when the initial call the following two methods)
 IOHIDManagerRegisterDeviceMatchingCallback(hid_mgr, &Handle_DeviceMatchingCallback, NULL);
IOHIDManagerRegisterDeviceRemovalCallback(hid_mgr, &Handle_DeviceRemovalCallback, NULL);
}
When the USB device is plugged into mac,This two methods will automatically call，So I think hidapi Why not consider this practice?
When you insert a device vid and pid meet the requirements, call get_device()，And When removing the equipment vid and pid meet the requirements, call remove_device().

amullins83, 2016-06-25:
Refer to what I said earlier. You're question to me sounds like this: "Mac OSX has this feature, so why not use it in HIDAPI?"

My answer is: the way this works on Windows and Linux is very different, to the point that we would potentially be breaking user's applications by forcing this feature to be turned on. The only way we could add a platform-independent callback feature like this would be to make it optional, and even then it simply wouldn't work as expected on every platform.

So, yes, it could be done, but I'm leaning toward it not being a great idea. I don't know. @signal11 probably has his own thoughts on the matter. I'm thinking the application code would need to do something like this:

    hid_init(); // Perform standard library initialization
    int error = hid_events_init(); // Turn on device events for the user's platform, which may fail.
    if (!error) {
      hid_events_register_insert_callback(my_hid_info, my_insert_callback); // Get registration data from an hid_info structure
      hid_events_register_remove_callback(my_hid_info, my_remove_callback);
    }

We would also need to add event system shutdown to the `hid_exit()` function.

jgouly, 2016-07-23:
I'm also interested in such a feature.
I have a UI, and I'd like to show when the device is connected or not, without polling hid_read()

asif-mahmud, 2016-07-28:
I did this using a thread to poll USB enumerate and read combination.
Didn't take much resource to do that.

>
> I'm also interested in such a feature.
> I have a UI, and I'd like to show when the device is connected or not,
without polling hid_read()

tyzthedoors, 2016-08-04:
@asif-mahmud I think we are in fact doing the extra work , when we are using polling hid_read() .Because the operating system is always in the scan usb .