signal11/hidapi #282: [LINUX] libudev breaking config

mikiemc73, 2016-07-10:
After running ./configure, I get a message:
Library libudev was not found on this system.
  Please install it and re-run ./configure

I have libudev-dev installed, plus libudev1.  Is there any way around this?

swt2c, 2016-07-11:
What does config.log show?

pine3825, 2017-09-30:
Do you have any idea??
==============================
jaehoon@jaehoon-UP:~/Workspace/ReSpeaker/respeaker-xmos-hid/hidapi$ ls
aclocal.m4      config.h.in    HACKING.txt       LICENSE.txt  pc
android         config.h.in~   hidapi            linux        README.txt
ar-lib          config.log     hidtest           ltmain.sh    stamp-h1
AUTHORS.txt     config.status  install-sh        m4           testgui
autom4te.cache  config.sub     libtool           mac          udev
bootstrap       configure      libusb            Makefile     windows
compile         configure.ac   LICENSE-bsd.txt   Makefile.am
config.guess    depcomp        LICENSE-gpl3.txt  Makefile.in
config.h        doxygen        LICENSE-orig.txt  missing
jaehoon@jaehoon-UP:~/Workspace/ReSpeaker/respeaker-xmos-hid/hidapi$ sudo apt-get install autotools-dev autoconf libtool
E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?

swt2c, 2017-10-01:
@pine3825 are you running Ubuntu?  If so, is Software Updater running?  If so, close it and try again.