signal11/hidapi #283: Support for Windows UWP apps

triplef, 2016-07-15:
hidapi on Windows currently uses the SetupApi library, which is not supported for Universal Windows Platform (UWP) apps. It seems that these calls need to be be replaced by functions from CfgMgr32, as described in this document:
https://msdn.microsoft.com/en-us/library/windows/hardware/mt210943(v=vs.85).aspx

Are there already any plans for this or any forks that might support UWP?

amullins83, 2016-07-15:
We probably need to split the Windows implementation into two versions, sort of like how we have libusb and hidraw implementations for Linux.

andrewputrajaya, 2017-03-02:
Even with CfgMgr32, in UWP you still cannot communicate with HID device with Microsoft driver. Only custom HID is accessible in UWP and it can be done easily using their built-in API (https://docs.microsoft.com/en-us/uwp/api/Windows.Devices.HumanInterfaceDevice). I tried it before using Raspberry Pi with Win 10 IoT core, but it didn't work. So ended up using Raspbian and hidapi.