signal11/hidapi #285: segfault on USB disconnect

sirop, 2016-08-13:
Hallo.

When disconnecting my USB joystick I get this as `dmesg` output:
````
[Sat Aug 13 10:01:03 2016] usb 1-3: USB disconnect, device number 20
[Sat Aug 13 10:01:05 2016] bodnar_USB_gene[6318]: segfault at 4 ip 00007fa98241d4d0 sp 00007fffcab38c78 error 4 in libhidapi-hidraw.so.0.0.0[7fa98241b000+3000]
```

Trying then to 
```
 hid_close(handle);
 hid_exit();
 hid_init();
 handle = hid_open(0x16c0, 0x05ba, serialNumberW);
```
does not bring anything then.

I tried it out also with libusb:
```
[Fri Aug 12 02:59:55 2016] usb 1-3: USB disconnect, device number 11
[Fri Aug 12 02:59:56 2016] bodnar_USB_gene[3113]: segfault at 24 ip 00007f021770a310 sp 00007fff739c5988 error 4 in libhidapi-libusb.so.0.0.0[7f0217706000+6000]
````

signal11, 2016-08-13:
Can you debug into it? Can you try with the latest version from github (git clone ...) ?

sirop, 2016-08-13:
I seem to have the latest github version:
```
master@debian-master:~/hidapi$ git log | head -10
commit b5b2e1779b6cd2edda3066bbbf0921a2d6b1c3c0
Author: Alan Ott <alan@signal11.us>
Date:   Thu Mar 3 10:54:51 2016 -0500

    windows: Always open with sharing enabled

    In the past, there was desire to make hid_open() open devices
    exclusively, preventing devices from being opened more than once.
    Unfortunately, recent versions of Windows have made this largely
    impossible for many devices, as it appears that Windows itself is
```

Maybe my xenomai kernel is to blame?
```
master@debian-master:~/hidapi$ uname -a
Linux debian-master 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1jessie~1da x86_64 GNU/Linux
```
I do not get any core dump if you meant that with debugging.

Does it matter if not only XHCI but also EHCI is enabled in BIOS?

sirop, 2016-08-13:
But I could do `gdb -p `:
```
(gdb) c
Continuing.

Program received signal SIGSEGV, Segmentation fault.
hid_read (dev=0x0, data=0x7fff0ac65c04 "c 0000 0000 300f 6630 3620", length=256) at hid.c:724
724             return hid_read_timeout(dev, data, length, (dev->blocking)? -1: 0);
(gdb) p dev
$1 = (hid_device *) 0x0
```
So seems that the device handle is set to `NULL` on disconnect?
But then these lines https://github.com/signal11/hidapi/blob/master/linux/hid.c#L699-L703
are a bit misleading as one get down there only with some luck.

signal11, 2016-08-13:
Then you are setting the device handle to NULL somehow. The library doesn't magically alter your pointers for you. You are passing NULL into hid_read() (see dev=0x0)

signal11, 2016-08-13:
If you want to break out the big hammer, put a data breakpoint on your handle, and you'll see when it changes, but I think you can probably find it with static analysis.

sirop, 2016-08-13:
My code is rather simple:
```     
   static hid_device *handle=NULL;
   handle = hid_open(0x16c0, 0x05ba, serialNumberW);
   hid_set_nonblocking(handle, 1);
         while (1) {
                res = hid_read(handle, buffer, sizeof(buffer)); 
        
                   if (res < 0) {
                     rtapi_print_msg(RTAPI_MSG_INFO, "HIDRAW: unable to read\n");
                     USBcomm = 0;
                     disconnCounter++;
                     if (disconnCounter >= 2){
                         hid_close(handle);
                         hid_exit();
                         hid_init();
                         handle = hid_open(0x16c0, 0x05ba, serialNumberW);
                         if (!handle) {
                             rtapi_print_msg(RTAPI_MSG_ERR, "Unable to open device\n");
                         } 
                         else {
                              hid_set_nonblocking(handle, 1);
                        }
                     }
                  }
                  else{
                           char swap[18];
                           USBcomm = 1;
                           disconnCounter = 0;
                           // parse the USB datagram
                  }
        usleep(25*1000);
   }
````
It means that you are right: ` handle = hid_open(0x16c0, 0x05ba, serialNumberW)` called in the loop returns `NULL` which is passed to `hid_read` in the next iteration.

closed 2016-08-13 by sirop