signal11/hidapi #287: Cannot build 0.7.0 with Ubuntu 16.04.1

frankiesimon, 2016-08-28:
With Ubuntu 16.04.01, following the instructions of running `make` from the linux subfolder.

First I got the error about libusb, which I then installed with sudo apt-get install libudev-dev.

But now the following error happens:

`g++ -Wall -g  hid-libusb.o ../hidtest/hidtest.o 'pkg-config libusb-1.0 libudev --libs' -o hidtest`
`/usr/bin/ld: hid-libusb.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'`
`//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line`
`collect2: error: ld returned 1 exit status`
`Makefile:25: recipe for target 'hidtest' failed`
`make: *** [hidtest] Error 1`

signal11, 2016-08-28:
Use the latest from git.

swt2c, 2016-08-29:
Or just use the Ubuntu packages.

bw1git, 2016-10-04:
I have the same problem.
I am also using the latest hidapi 0.7.0 with Ubuntu 16.04.01.
There is NO Ubuntu hidapi package. ( ONLY GITHUB )

I also already installed: (sudo apt-get install )
libusb-1.0
libusb-1.0-0-dev

sudo apt-get install autotools-dev autoconf automake libtool

Shows all files already loaded.

Any other suggestions?

signal11, 2016-10-04:
Don't use 0.7.0. Use the latest from git  (git clone ...)

swt2c, 2016-10-04:
There *are* Ubuntu packages:

libhidapi-dev is the development package
libhidapi-hidraw0 is the hidraw backend library
libhidapi-libusb0 is the libusb backend library