signal11/hidapi #289: hidapi on windows - undefined reference to functions

shan-94, 2016-09-01:
Hi. I build the hidapi on windows using Visual Studios. The build generated a 'hidapi.lib' file. Now I am trying to write C code using Eclipse IDE and have included the 'hidapi.lib' by changing it's name to 'libhid.lib'. The hidapi.h is also present and included. The error comes when I try to access any of the functions of the hidapi.h
The compiler gives an undefined reference error everytime I run the program. Where am  Igoing wrong ?

signal11, 2016-09-01:
You have to link to the library.

http://stackoverflow.com/questions/8332460/how-do-i-include-a-statically-linked-library-in-my-eclipse-c-project

Or you could build hid.c directly indo your project.

shan-94, 2016-09-02:
Do you mean linking the .lib file ? I have linked the hidapi.lib which was generated after building the project in MS VS. However, I have changed it's name to libhid.api. 
Is there any other file I am suppose to link ?
And how do I build hid.c directly into my project ?

shan-94, 2016-09-03:
Hi ... I tries building a project with MS VS and it worked fine without any errors.. As far as I am concerned, the problem is the incompatibility of the .lib file generated between visual studio and gcc

signal11, 2016-09-03:
Is hid_init() part of the version that you have? Make sure you're using the latest version from git (git clone ...).

Why don't you just build hid.c directly into your application?

There's no reason with an open source library to build it with a different compiler than you use for the application.