signal11/hidapi #292: Retro Joystick Adapter 3.0 hid_get_feature_report() not working under Windows

nyholku, 2016-09-08:
Hi,

of course this maybe something to do with the device itself but "Retro Joystick Adapter 3.0" the HIDAPI test/example application is unable to receive a feature report with hid_get_feature_report() under Windows but the same code under Mac OS X works just fine.

The device as seen by Mac OS X USB Prober looks like this:
 
http://www.sparetimelabs.com/temp/pic3.png

Incidentally I checked what the GetLastError returned after the failed    

`DeviceIoControl(dev>device_handle,IOCTL_HID_GET_FEATURE,`

call and the error code was 1.

I should probably own up that I run Windows 7 under Parallels Desktop virtual machine on my Mac OS X.

If you think this related to my circumstance or the device and don't want to investigate this further I apologise for the noise.

Here is what is see in Windows (under mingw shell):

```nyholku@NYHOLKU8525 /y/Downloads/hidapi-master/windows
$ hidtest.exe
Device Found
  type: 16c0 0a99
  path: \\?\hid#vid_16c0&pid_0a99#7&68b5d05&0&0000#{4d1e55b2-f16f-11cf-88cb-0011
11000030}
  serial_number: 000000
  Manufacturer: retronicdesign.com
  Product:      Retro Joystick Adapter v3.0
  Release:      100
  Interface:    -1

Device Found
  type: 203a fffb
  path: \\?\hid#vid_203a&pid_fffb#7&254eab45&0&0000#{4d1e55b2-f16f-11cf-88cb-001
111000030}
  serial_number: PW3.0
  Manufacturer: Parallels
  Product:      Virtual Keyboard
  Release:      100
  Interface:    -1

Device Found
  type: 203a fffc
  path: \\?\hid#vid_203a&pid_fffc&mi_00#7&37099d97&0&0000#{4d1e55b2-f16f-11cf-88
cb-001111000030}
  serial_number: PW3.0
  Manufacturer: Parallels
  Product:      Absolute Coordinate Interface
  Release:      100
  Interface:    0

Device Found
  type: 203a fffc
  path: \\?\hid#vid_203a&pid_fffc&mi_01#7&2de99099&0&0000#{4d1e55b2-f16f-11cf-88
cb-001111000030}
  serial_number: PW3.0
  Manufacturer: Parallels
  Product:      Relative Coordinate Interface
  Release:      100
  Interface:    1

Manufacturer String: retronicdesign.com
Product String: Retro Joystick Adapter v3.0
Serial Number String: (48) 000000
Indexed String 1: retronicdesign.com
Unable to send a feature report.
Unable to get a feature report.
Incorrect function.Unable to write()
Error: The parameter is incorrect.
Unable to write() (2)
waiting...
waiting...
waiting...
waiting...
```

mat1992, 2017-05-26:
Hi nyholku,

 I have exactly the same problem as you ! I work with windows 10, and I can't get or send a feature report, and also unable to write. But I can read all the other information of the device.
 Did you solve the problem ? 
 Thanks by advance.

nyholku, 2017-05-26:
Nope I did not. But I did not pursue this as signali11/HIDAPI is main 'target' I was just using it as a reference for m PureJavaHIDAPI.

mat1992, 2017-05-26:
Okay.
 Thanks a lot. 

 Does anyone can help me with this problem ?

ulao, 2018-01-31:
Old post but I found the same thing... the fix is to remove winusb and install hidusb back. Use zadig to see what you are currently using.

nyholku, 2018-01-31:
Hi ulao, thanks for reporting, this is sure to help someone.