signal11/hidapi #293: hid_get_feature_report() returns error

Moonbeem, 2016-10-13:
Hi,
My test program is unable to receive a feature report with hid_get_feature_report() under Linux but hid_send_feature_report() seems to work just fine.

`....................................
handle = hid_open(cur_dev->vendor_id, cur_dev->product_id, NULL);
memset(buf,0x00,sizeof(buf));
buf[0] = 0x55;
buf[1] = 0xf7;
buf[2] = 0xb3;
buf[3] = 0x02;
buf[4] = 0xa8;
buf[5] = 0xbb;
buf[6] = 0x0e;
res = hid_send_feature_report(handle, buf, 7);
if (res < 0)         printf("Unable to send a feature report.\n");
memset(buf,0x00,sizeof(buf));
buf[0] = 0x55;
res = hid_get_feature_report(handle, buf, sizeof(buf));
if (res < 0) {
                        printf("Unable to get a feature report.\n");
                        printf("%ls", hid_error(handle));
}
.................................................
`
error code:    ioctl(GFEATURE): No such device
                           Unable to get a feature report.

The HID device supports feature reports. I really need some help, and thanks very much!

signal11, 2016-10-13:
Byte buf[0] is the report number, or 0x0 if you're not using numbered reports.

Moonbeem, 2016-10-13:
Thank you for your reply. I am sure that 0x55 is the report number.

 
&mdash;

signal11, 2016-10-13:
I don't know then. Maybe the device doesn't support getting the feature report. Try calling it with the proper size (7). I assume this is a recent kernel.

Moonbeem, 2016-10-14:
Thanks. I will try it later.

closed 2016-11-01 by Moonbeem