signal11/hidapi #294: hid_open fails on Windows 8

Moonbeem, 2016-10-20:
Hi,
I'm having a similar issue in #247 . Could you let me kown the solution?

signal11, 2016-10-20:
You're trying to open a keyboard? HIDAPI doesn't open keyboards or mice on Windows. It's for custom HID devices.

Moonbeem, 2016-10-21:
I'm using touch screen HID devices.

signal11, 2016-10-21:
So in what way is it like 247?

Moonbeem, 2016-10-21:
I tried to open my device by VID and PID, but hid_open() always failed with a NULL handle. By debugging into the source, I found that CreateFileA() always returns INVALID_HANDLE_VALUE.
[image]

signal11, 2016-10-21:
Use the latest from git (git clone ...)

The touchscreen will report as a HID mouse most likely, preventing Windows from letting you open it.

Moonbeem, 2016-10-21:
Thank you for your patient.

Doraemoe, 2017-02-12:
Does HIDAPI also cannot open keyboards on Mac? I have 4 HID devices listed in my Mac. I can use `hid_open` to open Magic Trackpad 2 and Apple Internal Keyboard / Trackpad, but cannot open a usb keyboard and a bluetooth game controller.

signal11, 2017-02-16:
I have no idea. It's pretty much at the mercy of the OSX platform libraries and Apple likes to change stuff. HIDAPI is not for keyboards and mice anyway.

Boscop, 2017-11-16:
Why can't hidapi open keyboards? This is a bad limitation :/

(I want to open my Corsair K95 keyboard to get the keystrokes from the additional G keys on it. https://github.com/meh/rust-hid/issues/6)

tresf, 2017-11-16:
@Boscop I think you're off-topic and also, it's in the readme... 

```
Linux/hidraw (linux/hid.c):
This back-end uses the hidraw interface in the Linux kernel.  While this
back-end will support both USB and Bluetooth, it has some limitations on
kernels prior to 2.6.39, including the inability to send or receive feature
reports.  In addition, it will only communicate with devices which have
hidraw nodes associated with them.  Keyboards, mice, and some other devices
which are blacklisted from having hidraw nodes will not work. Fortunately,
for nearly all the uses of hidraw, this is not a problem.
```

Boscop, 2017-11-16:
@tresf I'm on Windows 8.1 and hid_open() fails with "not found" error. This IS a problem ;)

Btw, can you recommend a different HID lib that can open keyboards?

tresf, 2017-11-16:
@Boscop I'm sorry I made a mistake from the perspective that you were on topic.

Anyway, here's what the internet says about this problem (note, it's a link back to this tracker).

https://github.com/signal11/hidapi/issues/247#issuecomment-161152387

I'll quote the reply:

> Security. It's a security problem if a normally-privileged application can keystroke-log the system keyboard. Windows won't allow you to open a keyboard with hidd.dll. If you find a way, let me know.
>
> HIDAPI is for custom-class HID devices, devices which use HID to transfer arbitrary data to a PC.