signal11/hidapi #304: hid_enumerate returns NULL when running in VirtualBox (host: Mac OSX; guest: Ubuntu 16.04)

kennethjiang, 2016-12-19:
As in the title, I tried hid_enumerate in a VirtualBox VM and it always return NULL

A simple test code is as follows:
```c
#include <stdio.h>
#include <hidapi/hidapi.h>
int main(int args, char** argv) {
	struct hid_device_info *devs = hid_enumerate(0x0, 0x0);
	if (!devs) {
		printf ("not found!");
	}
}
```

However `lsusb` works:
```bash
ubuntu@ubuntu-xenial:~$ lsusb
Bus 001 Device 002: ID 03eb:2111 Atmel Corp.
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

This problem is preventing [openocd](http://openocd.org/) from running in a Ubuntu 16.04 VirtualBox VM (host is Mac OSX). The USB device is mapped to guest VM through VirutalBox.

I tested 2 HIDAPI version: 0.7.0 and master on github. Both versions have the same problem.

signal11, 2016-12-19:
It means no HID devices are connected.

kennethjiang, 2016-12-19:
hmmm, I assume since `lsusb` returns the correct devices, they should be discoverable by HIDAPI.

Also when i ran exactly the same test program in an Ubuntu on bare metal, it worked.

What am I missing here?

kennethjiang, 2016-12-19:
Ah, it looks like not all usb devices are presented as hid devices on linux. Will dig in a bit more to see went wrong.

kennethjiang, 2016-12-20:
It looks like /dev/usb is missing from VirtualBox vm for some unknown reason. closing this issue.

closed 2016-12-20 by kennethjiang