signal11/hidapi #305: hidapi-libusb don't reattach kernel driver

Milek7, 2016-12-29:
hidapi with libusb backend detach device kernel driver in hid_open, but don't reattach it in hid_close. It needs to either call libusb_attach_kernel_driver or use libusb_set_auto_detach_kernel_driver to make libusb automatically detach/attach driver in libusb_claim_interface/libusb_release_interface.

signal11, 2016-12-29:
And what happens when the application crashes or otherwise exits uncleanly? There's no way to ensure that it's ever actually done.

HIDAPI is for custom devices. If you have a custom device that you made that you're making the software for, then the default kernel driver won't be of much help.

seysn, 2018-04-09:
Hello,

I encounter the same problem, my mouse and my keyboard are stuck when I try to use `hid_open` with libusb.
I think this is because, as said before, the device is not reattached to the kernel.
With these 2 lines at the beginning of `hid_close` I can regain control of my devices when running hidtest :

```c
libusb_release_interface(dev->device_handle, dev->interface);
libusb_attach_kernel_driver(dev->device_handle, dev->interface);
```

Is it intended that you do not reattach the device to the kernel at any moment in the code ?
Also, since I don't want my devices to be stuck, is there a better way to reattach my device other than these two lines in the start of `hid_close`?