signal11/hidapi #309: TI MSP430 FR5994 on OSX shows up with ioreg but not with hidtest

petertorelli, 2017-01-24:
Hi there. My issue is similar to a few issues I've spotted in this repo (like #306 and kinda #301). #306 is the closest issue but there's no reply to it. I am unable to see my device -- MSP Tools Driver ---  after building the (latest) hidtest on OSX 10.12.2. I can see it with ioreg -p IOUSB. I have several other TI devices that use this interface, they also exhibit the same issue.

```
% ioreg -p IOUSB 
+-o Root  <class IORegistryEntry, id 0x100000100, retain 15>
  +-o Root Hub Simulation Simulation@14000000  <class AppleUSBRootHubDevice, id 0x100000301, registered, matched, active, busy 0 (511 ms), retain 15>
    +-o BRCM20702 Hub@14300000  <class AppleUSBDevice, id 0x10000d66a, registered, matched, active, busy 0 (17 ms), retain 20>
    | +-o Bluetooth USB Host Controller@14330000  <class AppleUSBDevice, id 0x10000d6b2, registered, matched, active, busy 0 (1159 ms), retain 25>
    +-o Apple Internal Keyboard / Trackpad@14400000  <class AppleUSBDevice, id 0x10000d67d, registered, matched, active, busy 0 (95 ms), retain 23>
    +-o Goldtouch USB Keyboard@14100000  <class AppleUSBDevice, id 0x10000d7bd, registered, matched, active, busy 0 (79 ms), retain 21>
    +-o MSP Tools Driver@14200000  <class AppleUSBDevice, id 0x100014f85, registered, matched, active, busy 0 (46 ms), retain 26>
```

But hidtest shows different results, without the the device:

```
% ./hidtest 
Device Found
  type: 05ac 0262
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS12@14400000/Apple Internal Keyboard / Trackpad@14400000/Apple Internal Keyboard@0/AppleUSBTCKeyboard@14400000,0
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Release:      225
  Interface:    -1

Device Found
  type: 05ac 030d
  path: IOService:/IOResources/IOBluetoothHCIController/AppleBroadcomBluetoothHostController/IOBluetoothDevice/IOBluetoothL2CAPChannel/BNBMouseDevice
  serial_number: 58-1f-aa-fa-ac-93
  Manufacturer: Apple
  Release:      306
  Interface:    -1

Device Found
  type: 04ca 005b
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Goldtouch USB Keyboard@14100000/IOUSBHostInterface@1/IOUSBHostHIDDevice@14100000,1
  serial_number: 
  Manufacturer: Lite-On Technology Corp.
  Product:      Goldtouch USB Keyboard
  Release:      119
  Interface:    -1

Device Found
  type: 04ca 005b
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Goldtouch USB Keyboard@14100000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14100000,0
  serial_number: 
  Manufacturer: Lite-On Technology Corp.
  Product:      Goldtouch USB Keyboard
  Release:      119
  Interface:    -1

Device Found
  type: 05ac 0262
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS12@14400000/Apple Internal Keyboard / Trackpad@14400000/Touchpad@2/AppleUSBTCButtons@14400000,2
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Release:      225
  Interface:    -1

Device Found
  type: 05ac 0262
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS12@14400000/Apple Internal Keyboard / Trackpad@14400000/Touchpad@1/AppleUSBMultitouchDriver@14400000,1
  serial_number: 
  Manufacturer: Apple Inc.
  Product:      Apple Internal Keyboard / Trackpad
  Release:      225
  Interface:    -1

```
Advice?

Thanks,
Peter

mrpippy, 2017-01-24:
Can you post the output of `ioreg -Src IOHIDInterface`? Are you sure that "MSP Tools Driver" is a HID device?

petertorelli, 2017-01-24:
It does not show up as a HID device. That's the problem. I assumed that because the drivers the OEM sent me required the HID API automatically meant that it was a HID device. Thank you!

closed 2017-01-24 by petertorelli