signal11/hidapi #313: unable to open device

rsandoz, 2017-02-24:
I am trying to open a lighting strip (on a Razor Mousepad) vid=0x1532 pid=0x0c00 and it is failing at the CreateFile function (5 - access denied).  It is using the MS hid driver as far as I can tell (HidUsb (v10.0.14393.351)).  Is there something I am missing for opening this device?  I ultimately want to send control transfer messages to it to set the LEDs.

rsandoz, 2017-02-24:
FYI, I can communicate with it in this fashion when I use libusb-win32 with a filter-driver.

signal11, 2017-02-27:
You can't open keyboards or mice on Windows using the platform HID library.

rsandoz, 2017-03-02:
It is not a mouse or a keyboard.  Furthermore, I don't buy this "You can't" answer.  My rationale is as follows:  Razer Synapse can send control transfer messages to it.  I have observered this by physically using their application affect the lighting as well as use wireshark to observe the messages being sent from the host to the USB device when I use the Razer application.  How is the Razer end application able to send these control tranfer messages through this driver?

tresf, 2017-03-02:
> It is not a mouse or a keyboard.

I had misread this as a pointing device too...

It's an illuminated mouse pad and the LEDs are controllable.

[image]

Perhaps @pez2001 has some insight over at https://github.com/pez2001/razer_chroma_drivers.

rsandoz, 2017-03-02:
I initially observed all interfaces as follows:
Razer Firefly (Composite Parent) - usbccgp (v10.0.14393.0)
Razer Firefly (Interface 0) - HidUsb (v10.0.14393.351)
Razer Firefly (Interface 1) - HidUsb (v10.0.14393.351)
Razer Firefly (Interface 2) - HidUsb (v10.0.14393.351)
And with it like this can control it through the Synapse software.

Now if I go into zadig and replace the driver on Interface 0 with winusb:
Razer Firefly (Interface 0) - WinUSB (v6.1.7600.16385)
Not only does it disappear from the Razer Synapse software --> I am able to control it with my own C code.

```
#pragma comment(lib, "Winusb.lib")
#pragma comment(lib, "setupapi.lib")

#define RAZER_VID 0x1532
#define RBP2016_PID 0x0C00

HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\DeviceAccess\CapabilityHandlers\{DEE824EF-729B-4A0E-9C14-B7117D33A817} // winusb GUID
...
HDEVINFO hDevInfo = SetupDiGetClassDevs(&GUID_DEVINTERFACE, 0, 0, DIGCF_DEVICEINTERFACE);
...
HANDLE hDevice = CreateFile(pData->DevicePath
...
bResult = WinUsb_Initialize(hDevice, &hWinUSBHandle);
...
WINUSB_SETUP_PACKET packet;
...
WinUsb_ControlTransfer(hWinUSBHandle, packet, response, REPORT_SIZE, &cbSent, 0);
```

I don't want to have to install an additional driver.  I want to control it in the same way that the Razer software does vid the HidUsb driver.

Furthermore, this whole mouse and keyboard thing really seems inapplicable, as I am also able to reproduce this same paradigm on Razer keyboards and mice.  I don't have the source code to the Razer stuff, but it seems they are using these hidclass.sys hidparse.sys and hidusb.sys in some way shape or form to send these control transfer messages.

signal11, 2017-03-07:
When you say you "don't buy" the inability of HIDAPI to open mice and keyboards in Windows.... the Windows platform HID functions won't allow you to talk to them using those APIs. That's all there is to it, really.

Since this is not a mouse, as you point out, you should be able to open it with HIDAPI. I'd say to make sure you're opening the right interface. Use hid_enumerate() to list the hid interfaces on the system and try opening each of them. I don't know why it takes three interfaces to have LED lights, but it is what it is. See if that helps. Are you using the latest from git (ie: git clone ...)?

Alan.

rsandoz, 2017-03-09:
I am saying that Razer is sending messages to the mice and keyboard devices to control the LED lighting on these devices.  The only drivers that are installed on these devices are kbdclass.sys, kbdhid.sys, mouclass.sys, mouhid.sys, hidclass.sys, hidparse.sys, hidusb.sys.  When I open the application Razer Synapse and send commands to these LEDs, how is this being done if not through these drivers?  I can even monitor the messages with wireshark.

signal11, 2017-03-15:
Well my library sits on top of those platform libraries. If you find something missing that your device needs, let me know.

It's also possible that your device's software has some kind of service running that opens the device in exclusive mode, preventing me from opening it. Maybe try after uninstalling the software that comes with your device.

CalcProgrammer1, 2020-07-01:
Just going to chime in here to say that I've updated @rsandoz's Razer drivers port to use hidapi without the WinUSB driver.  I had to get the usage page correct and do some size tweaking on the feature report.  I also had to create a custom hidapi.h and rename the hid_device struct to hidapi_device because of a name conflict with the hid_device in the fake Linux kernel headers.

I'm able to control my entire Razer setup without WinUSB now.

https://github.com/CalcProgrammer1/openrazer-win32