signal11/hidapi #315: uninitialized strings on windows hid.c

JosepMaJAZ, 2017-03-05:
There is an unitialized memory in windows/hid.c that can cause garbage output (and maybe even crashes).
Basically, what has been detected is that on devices that do not report some of the HidD_Getxxxx() methods (i.e. that res = FALSE) is that random text appears on serial_number, manufacturer_string and product_string, basically because it's never filled.
Also, given that we also have:
		free(d->serial_number);
		free(d->manufacturer_string);
		free(d->product_string);
it could perfectly crash in this situation.

Additionally, I add wstr[0] = 0x0000 in case for whatever reason the call to the method returned TRUE but didn't touch the string.

I am not forking and creating a Pull Request, because seeing that a 2014 Pull request with a few lines, for which the project owner has accepted the fix, being still there open, doesn't give me much confidence of being worth the effort.

@392:
tmp = (struct hid_device_info*) calloc(1, sizeof(struct hid_device_info));
+++ // Initialize all memory to zero:
+++  memset(tmp, 0, sizeof(struct hid_device_info));

And then:

@425:
+++ wstr[0] = 0x0000;
/* Serial Number */
res = HidD_GetSerialNumberString(write_handle, wstr, sizeof(wstr));

rryan, 2017-03-05:
Are you sure the memset is needed? calloc both allocates and zeros.

JosepMaJAZ, 2017-03-05:
Mmmm... Now that I read the definition on cppreference, indeed, calloc should already be doing that.

What I can assert is that with these two changes, I got the problem fixed on Mixxx. Maybe it turns out it was simply the string initialization.

signal11, 2017-03-07:
Further, free(NULL) is perfectly valid in C.

JosepMaJAZ, 2017-03-07:
Ok, let's start again.

I was wrong with the "memset" requirement (and what it came out from that, namely the problems with free). If calloc is working as expected with visual studio, then that should not be the problem I was experiencing.

Now, there is the second change, the wstr[0] = 0x0000; 
It is expected that this buffer is filled with the string if the call to the Hid method returns TRUE. Well, in my case, for this particular hardware/driver, I was getting uninitialized data (random unicode characters that most of the time looked like chinese).
When i did the test with these two changes, i could get an empty string, which is all I was trying to achieve.

signal11, 2017-03-07:
Maybe the platform code is broken then. Can you do the same for all the strings and submit a patch?

JosepMaJAZ, 2017-03-07:
You mean, this?

diff --git a/windows/hid.c b/windows/hid.c
index 86810d7..9c7f82c 100755
--- a/windows/hid.c
+++ b/windows/hid.c
@@ -434,6 +434,7 @@ struct hid_device_info HID_API_EXPORT * HID_API_CALL hid_enumerate(unsigned shor
                        else
                                cur_dev->path = NULL;

+                       wstr[0] = 0x0000;
                        /* Serial Number */
                        res = HidD_GetSerialNumberString(write_handle, wstr, sizeof(wstr));
                        wstr[WSTR_LEN-1] = 0x0000;

signal11, 2017-03-08:
I mean submit it as a pull request, and not just for the serial number string, for all the strings. Also, put the added line after the comment (rather than before)