signal11/hidapi #319: after long time and then unable to open device

AnCruise, 2017-03-24:
hi, I'v used the lib in MFC to control a finger device. At the beginning it works well ,but when I stop it after the  hid_write() and hid_read() run 11 millions times (by Bus Hound),  and then I open my device, it report 'unable to open device'. Is there some wrong?
`bool COXiViewView::openHID(){
	...
	hid_init();
	devs = hid_enumerate(HID_VID,HID_PID);
	hid_free_enumeration(devs);
	handle = hid_open(HID_VID, HID_PID,SERIAL_NO);
	...
	if (!handle) {
		pEdt->ReplaceSel(L"unable to open device\r\n");	
		return false;
	}	
	else	return true;
}`
`bool COXiViewView::closeHID(){
	...
	hid_close(handle);
	if (0 != hid_exit())
	{
		pEdt->ReplaceSel(L"hid_exit error\r\n");	
		return false;
	}
	return true;
}`
`if (-1 == hid_write(handle,WriteTempBuffer,65))
	{
		strLog.Format(L"hid_write Error() = %s\r\n",hid_error(handle));
		pEdt->SetSel(-1,-1);
		pEdt->ReplaceSel(strLog);
	}
if (-1==hid_read_timeout(handle,ReadTempBuffer,65,1000))
{
		strLog.Format(L"hid_read_timeout Error() = %s\r\n",hid_error(handle));
		pEdt->SetSel(-1,-1);
		pEdt->ReplaceSel(strLog);
}`

closed 2017-03-27 by AnCruise

signal11, 2017-03-27:
Looks like you closed this yourself....

For what it's worth:
hid_enumerate() is not necessary in your case where you use hid_open().

Only call hid_init() once at the beginning of execution, not every time you open a device. Unless you're in some kind of multithreaded environment, you don't need to call it at all actually.

You say USB_BULK_BUFF, but it's not actually a bulk endpoint. It's an interrupt endpoint.

Alan.

AnCruise, 2017-03-29:
Thanks for your help , Alan .
I have resolved the problem . Yes , I used it in a multithreaded environment , so every time it is necessary to call it . And the USB_BULK_BUFF is a number I have defined , it is causing you to misunderstand , sorry.
William.