signal11/hidapi #321: unable to open device in rpi (jessie)

danielsulik, 2017-04-01:
Hi
first of all great lib and thank you for it.

I have custom HID device running on STM32L152 under windows 7.
The same device enumerates OK in jessie on rpi however when running hidtest
I am getting error: unabke to open device
lsusb dump below.

Any idea?
Thank you for you help.

Device Found
  type: 0458 003a
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Genius
  Product:      USB Optical Mouse
  Release:      100
  Interface:    0

Device Found
  type: 1c4f 0002
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: SIGMACHIP
  Product:      USB Keyboard
  Release:      110
  Interface:    0

Device Found
  type: 1c4f 0002
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: SIGMACHIP
  Product:      USB Keyboard
  Release:      110
  Interface:    1

Device Found
  type: 0483 5750
  path: /dev/hidraw3
  serial_number: B1544EABD546
  Manufacturer: STMicroelectronics
  Product:      Custome HID in FS Mode
  Release:      200
  Interface:    0

unable to open device

Bus 001 Device 010: ID 0483:5750 STMicroelectronics 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0483 STMicroelectronics
  idProduct          0x5750 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      25
         Report Descriptors: 
           ** UNAVAILABLE **
     Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

danielsulik, 2017-04-02:
HID descriptor below
#define HID_BUFFER_RAW_SIZE 64

__ALIGN_BEGIN static uint8_t CustomHID_ReportDesc[USBD_CUSTOM_HID_REPORT_DESC_SIZE] __ALIGN_END =
{
	0x06, 0x00, 0xff,          // USAGE_PAGE (Vendor Defined Page 1)
	0x09, 0x01,                // USAGE (Vendor Usage 1)
	0xa1, 0x01,                // COLLECTION (Application)
	0x15, 0x00,                //   LOGICAL_MINIMUM (0)
	0x26, 0xff, 0x00,          //   LOGICAL_MAXIMUM (255)
	0x75, 0x08,                //   REPORT_SIZE (8)
	0x95, HID_BUFFER_RAW_SIZE, //   REPORT_COUNT (64)
	0x09, 0x01,                //   USAGE (Vendor Usage 1)
	0x81, 0x02,                //   INPUT (Data,Ary,Abs)
	0x09, 0x01,                //   USAGE (Vendor Usage 1)
	0x91, 0x02,                //   OUTPUT (Data,Ary,Abs)
	0xc0                       // END_COLLECTION
};

signal11, 2017-04-03:
Have you looked at permissions on the /dev/hidraw* devices?

danielsulik, 2017-04-03:
> Odoslané: Monday, April 3, 2017 16:57
> Komu: signal11/hidapi <hidapi@noreply.github.com>
> Kópia: Daniel Sulik <dan@meta-designs.com>; Author <author@noreply.github.com>

Ah silly me,

Of course permissions, that was the issue.

Seems, I need a break ;-)

Thank you

Dan

ravkr, 2017-06-05:
Can we close that issue?

danielsulik, 2017-06-05:
Sure you can close it

closed 2017-06-05 by danielsulik