signal11/hidapi #322: segfault in Linux (libusb & hidraw)

jsuiker, 2017-04-04:
I'm trying to hack around with a Symbol (Motorola/Zebra) barcode scanner in Linux.  For whatever reason I can't get the sample code to work properly.  I'm getting a segfault on the third call into the library (the first one passing 'handle' as a parameter). I've tried with the libusb backend as well as hidraw.

    #include <stdio.h>
    #include <stdlib.h>
    #include "/usr/local/include/hidapi/hidapi.h"

    #define MAX_STR 255

    int main(int argc, char* argv[])
    {
    	int res;
    	unsigned char buf[65];
    	wchar_t wstr[MAX_STR];
    	hid_device *handle;
    	int i;

    	// Initialize the hidapi library
    	res = hid_init();

    	// Open the device using the VID, PID,
    	// and optionally the Serial number.
    	handle = hid_open(0x05e0, 0x1200, NULL);

    	// // Read the Manufacturer String
    	res = hid_get_manufacturer_string(handle, wstr, MAX_STR);

Here's the command I'm using to build the code:

    gcc ~/symbol/hidapi/hidapi.c -o ~/symbol/hidapi/hidapi -L/usr/local/include/hidapi -lhidapi-libusb -g

... and the gdb output I'm getting ...

    [Thread debugging using libthread_db enabled]
    Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
    [New Thread 0x7ffff71cb700 (LWP 12673)]

    Thread 1 "hidapi" received signal SIGSEGV, Segmentation fault.
    0x00007ffff7bd5c16 in hid_get_manufacturer_string () from /usr/lib/x86_64-linux-gnu/libhidapi-libusb.so.0

signal11, 2017-04-04:
Are you checking that handle is not null? From gdb type "print handle".

jsuiker, 2017-04-05:
I suppose I have to put a break point in there first? looks like it IS null, but what can I do about that?  I first tried running with the default kernel driver (in HID keyboard mode and HID raw mode) and I've also tried using libusb to detach the kernel driver before running the above code...

I'm pretty new to this all I'm sure I need a better conceptual model of what's going on here...

signal11, 2017-04-05:
It's null because it can't open the device.
   https://github.com/signal11/hidapi/blob/master/hidapi/hidapi.h#L144

Is it permissions? Look in the udev/ directory in the hidapi source.

jsuiker, 2017-04-05:
oh wow, how simple.  thanks!

... now on to actually debugging the HID interface :+1:

closed 2017-04-05 by jsuiker