signal11/hidapi #323: does not enumerate any usb devices at all

flok99, 2017-04-11:
Kernel: 3.4.103-00033-g9a1cd03-dirty (olimex a20) or 4.9.20-v7+ (raspberry pi 3)
devices:
root@A20-OLinuXino:/usr/src/clewarecontrol# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 006: ID 0d50:0020 Cleware GmbH 
Bus 004 Device 005: ID 0d50:0030 Cleware GmbH 
Bus 004 Device 004: ID 0d50:0030 Cleware GmbH 
Bus 004 Device 003: ID 0d50:0008 Cleware GmbH 
Bus 004 Device 002: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

test-program:

#include <stdio.h>
#include <hidapi/hidapi.h>

int main(int argc, char *argv[])
{
        if (hid_init() == -1)
                fprintf(stderr, "Error initializing HID library\n");

        struct hid_device_info *cur_dev = hid_enumerate(0, 0);

        while (cur_dev)
                printf("%04x %04x\n", cur_dev->vendor_id, cur_dev->product_id);

        return 0;
}

This program returns no (none, zero) devices.

signal11, 2017-04-17:
Are any of those HID devices?

flok99, 2017-04-17:
Yes (the cleware ones).

signal11, 2017-04-17:
Are you using the hidraw implementation or the libusb implementation? Can you try the other one? Can you step into it with a debugger? You haven't given me much to go on.

flok99, 2017-04-18:
Hi,

With the libusb variant it all runs fine, it is the hidraw version which is broken. I was told by the libusb guys not to use it for hid-devices.
I'll see later if I can debug it.

flok99, 2017-04-18:
udev_enumerate_new returns 0x1423038
udev_enumerate_add_match_subsystem returns 0
udev_enumerate_scan_devices returns 0
and udev_enumerate_get_list_entry returns NULL <--

signal11, 2017-04-18:
> With the libusb variant it all runs fine, it is the hidraw version which is broken.

Some devices are blacklisted for hidraw in the kernel.

> I was told by the libusb guys not to use it for hid-devices.

Yes, don't use it directly. Use hidraw-libusb.

Alan.

flok99, 2017-04-18:
On Tue, Apr 18, 2017 at 6:05 PM, Alan Ott <notifications@github.com> wrote:

> With the libusb variant it all runs fine, it is the hidraw version which
> is broken.
>
> Some devices are blacklisted for hidraw in the kernel.
>

This range of devices is not (checked it with 4.11-rc7).

> I was told by the libusb guys not to use it for hid-devices.
>
> Yes, don't use it directly. Use hidraw-libusb.
>

Ok will do.
Thanks.

But I'm still curious why the hidraw version does not work.

signal11, 2017-04-18:
It's a good question. Step through the code and see what you find. Also see what's in /dev/input/hidraw* . Maybe it's permissions related. See files in the udev/ directory in the HIDAPI source.

jsuiker, 2017-04-28:
Guys,
there's a lot of info here and I'm sure others reading would love some explanations :)

1. Where can we find this magical list of disallowed hidraw devices?
2. What is this hidraw-libusb thing? I guess you're not talking about the libusb implementation of hidapi...
3. When should we 'not use hidraw for hid-devices' ...It sounds like a bit of a blanket statement but I don't understand why it was made

My own experimentation with the hidraw implementation of this library (hidapi) is having some trouble. I am using the testgui app on Ubuntu 16.10 and wondering why I can connect to devices when using hidapi-libusb-testgui but get an error message when I try using the hidapi-hidraw-testgui on the same devices (a barcode scanner and a mag-strip reader)

signal11, 2017-04-28:
> Where can we find this magical list of disallowed hidraw devices?

Start here: http://lxr.free-electrons.com/source/drivers/hid/usbhid/hid-quirks.c

> What is this hidraw-libusb thing? I guess you're not talking about the libusb implementation of hidapi...

Sorry, I meant hidapi-libusb. It's the HIDAPI implementation which is implemented using libusb underneath it.

> When should we 'not use hidraw for hid-devices' ...It sounds like a bit of a blanket statement but I don't understand why it was made

What I thought you meant was that the libusb people said not to use libusb (directly) for HID devices (which is one thing they (and I) do say). So...

Don't use libusb directly to talk to HID devices.
Use HIDAPI to talk to HID devices
HIDAPI has two implementations, one based on Linux hidraw, and one based on libusb. Use either, depending on what works for you. Each has limitations. The libusb implementation obviously won't do bluetooth, where hidraw can. Some devices are blacklisted in hidraw, so either patch the kernel to unblacklist them or use hidapi-libusb.

I hope this helps clear it up.

Alan.

jsuiker, 2017-04-28:
Awesome, thanks for the reference... I see that one of the devices I'm having trouble with is listed there. Looks like the HID_QUIRK_NOGET flag is set for it on line 137.  The other, however, is not listed.  So there's still something the matter with my MagTek MSR.  Let me know if you think a new 'issue' is warranted here.

signal11, 2017-04-28:
This is the first you mention of MagTek MSR in this issue.

jsuiker, 2017-04-29:
I created a new issue since I'm not having trouble enumerating devices... #325