signal11/hidapi #325: hidraw testgui not connecting

jsuiker, 2017-04-29:
I am trying to get a few hid peripherals to work with this library (and my linux box generally). After having some trouble, I decided to try out the testgui to see if I could do some baseline tests.  The libusb version of the testgui works just fine, I can enumerate and connect to the devices I'm working with.  Using the hidraw version, I cannot.  The devices are listed but clicking connect throws an error.  Unfortunately the error is not very descriptive, just something to the effect of 'unable to connect'

Thanks to a previous ticket, @signal11 helped me figure out how to add udev rules and whatnot.

Also thanks to @signal1, I noticed that one of the devices, a Symbol barcode scanner, has the [HID_QUIRK_NOGET](http://lxr.free-electrons.com/source/drivers/hid/usbhid/hid-quirks.c#L137) flag set for it. I assume that means I'll have to write some code for it but I'm not quite sure what I'll have to do just yet.  In any case, the other device I am using behaves the same way but does not have any flags set.

signal11, 2017-04-29:
Are the permissions for the device wrong? look at dmesg and the files in /dev/input/hidraw* .

jsuiker, 2017-04-29:
I'm using the non-keyboard version of the scanne.  I have both versions, one is an hid input device (keyboard emulator) and one is a "raw" (?) hid device.  Interestingly enough, I just plugged in the keyboard emulator version to see if it was any different... turns out it works just fine with the hidraw testgui while the "raw" device doesn't work.

the first
```
[13190.306193] usb 1-1.6: new full-speed USB device number 3 using ehci-pci
[13190.417511] usb 1-1.6: New USB device found, idVendor=0801, idProduct=0001
[13190.417513] usb 1-1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13190.417514] usb 1-1.6: Product: USB Swipe Reader
[13190.417515] usb 1-1.6: Manufacturer: Mag-Tek
[13190.419645] input: Mag-Tek USB Swipe Reader as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.6/1-1.6:1.0/0003:0801:0001.000F/input/input26
[13190.478456] hid-generic 0003:0801:0001.000F: input,hidraw6: USB HID v1.00 Keyboard [Mag-Tek USB Swipe Reader] on usb-0000:00:1a.0-1.6/input0
```
the second
```
[13462.982891] usb 2-1.6.4.1: USB disconnect, device number 24
[13466.512775] usb 2-1.6.4.1: new full-speed USB device number 25 using ehci-pci
[13466.629384] usb 2-1.6.4.1: New USB device found, idVendor=0801, idProduct=0002
[13466.629386] usb 2-1.6.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13466.629387] usb 2-1.6.4.1: Product: USB Swipe Reader
[13466.629388] usb 2-1.6.4.1: Manufacturer: Mag-Tek
[13466.632943] hid-generic 0003:0801:0002.0016: hiddev0,hidraw8: USB HID v1.00 Device [Mag-Tek USB Swipe Reader] on usb-0000:00:1d.0-1.6.4.1/input0
```

jsuiker, 2017-05-03:
is there any other debug info I could provide? I'd love to figure this out...

signal11, 2017-05-03:
You didn't try all the things I suggested yet.

jsuiker, 2017-05-03:
```
/dev/input$ ll
total 0
drwxr-xr-x  4 root root     520 May  3 12:46 ./
drwxr-xr-x 20 root root    4680 May  3 12:46 ../
drwxr-xr-x  2 root root     200 May  3 12:46 by-id/
drwxr-xr-x  2 root root     200 May  3 12:46 by-path/
crw-rw----  1 root input 13, 64 May  3 12:45 event0
crw-rw----  1 root input 13, 65 May  3 12:45 event1
crw-rw----  1 root input 13, 74 May  3 12:45 event10
crw-rw----  1 root input 13, 75 May  3 12:45 event11
crw-rw----  1 root input 13, 76 May  3 12:45 event12
crw-rw----  1 root input 13, 77 May  3 12:45 event13
crw-rw----  1 root input 13, 78 May  3 12:45 event14
crw-rw----  1 root input 13, 79 May  3 12:45 event15
crw-rw----  1 root input 13, 80 May  3 12:46 event16
crw-rw----  1 root input 13, 81 May  3 12:46 event17
crw-rw----  1 root input 13, 82 May  3 12:46 event18
crw-rw----  1 root input 13, 66 May  3 12:45 event2
crw-rw----  1 root input 13, 67 May  3 12:45 event3
crw-rw----  1 root input 13, 68 May  3 12:45 event4
crw-rw----  1 root input 13, 69 May  3 12:46 event5
crw-rw----  1 root input 13, 70 May  3 12:46 event6
crw-rw----  1 root input 13, 71 May  3 12:46 event7
crw-rw----  1 root input 13, 72 May  3 12:46 event8
crw-rw----  1 root input 13, 73 May  3 12:45 event9
crw-rw----  1 root input 13, 63 May  3 12:45 mice
crw-rw----  1 root input 13, 32 May  3 12:45 mouse0
crw-rw----  1 root input 13, 33 May  3 12:46 mouse1
```
contents of /etc/udev/rules.d/magtek.rules
```
# HIDAPI/libusb
SUBSYSTEM=="usb", ATTRS{idVendor}=="0801", ATTRS{idProduct}=="0001", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0801", ATTRS{idProduct}=="0002", MODE="0666"

# HIDAPI/hidraw
KERNEL=="hidraw*", ATTRS{busnum}=="1", ATTRS{idVendor}=="0801", ATTRS{idProduct}=="0001", MODE="0666"
KERNEL=="hidraw*", ATTRS{busnum}=="1", ATTRS{idVendor}=="0801", ATTRS{idProduct}=="0002", MODE="0666"
```

signal11, 2017-05-03:
Is your user in the input group? Also it looks like your rules file is having no effect.

signal11, 2017-05-03:
Did you notice that you have no hidraw* nodes in that folder. Are they in /dev?

signal11, 2017-05-03:
try taking out busnum in the udev file.

jsuiker, 2017-05-03:
> Did you notice that you have no hidraw* nodes in that folder. Are they in /dev?

They were indeed... and the permissions weren't set properly on the "raw" device

> try taking out busnum in the udev file.

That seems to have done the trick for the permissions... aaaaaand the sample application works now....

jsuiker, 2017-05-03:
So what's all that mess about HID_QUIRK_NOGET? I got the impression that it would disable functionality in HIDRAW mode for my other device?

closed 2017-05-03 by jsuiker