signal11/hidapi #328: hello,I'm a student from China;  I have a problem do not know how to fix it ;

zhoulilidamowang, 2017-05-08:
macOS sierra 10.12.3 
coreI5 2.7GHz
8GB MHz ddr3
ssd 256GB

CODE:
INFO_LOG()<<"hid_write begin";
ret = hid_write(m_pUsbDev, buf, 1025);
INFO_LOG()<<"hid_write end"

Log:
2017-05-08 17:37:12,468(-1330229248)[ INFO]{Write} hid_write begin
2017-05-08 17:37:13,476(-1330229248)[ INFO]{Write} hid_write end

2017-05-08 17:37:13,476(-1330229248)[ INFO]{Write} hid_write begin
2017-05-08 17:37:14,486(-1330229248)[ INFO]{Write} hid_write end

2017-05-08 17:37:14,486(-1330229248)[ INFO]{Write} hid_write begin
2017-05-08 17:37:15,496(-1330229248)[ INFO]{Write} hid_write end

2017-05-08 17:37:15,496(-1330229248)[ INFO]{Write} hid_write begin
2017-05-08 17:37:16,506(-1330229248)[ INFO]{Write} hid_write end

question:

Why send a buf takes 1 second?
But there is no problem under windows 7

signal11, 2017-05-08:
Sending is synchronous. Maybe your device is slow to accept the data.

zhoulilidamowang, 2017-05-08:
But my device in the win7 can be used normally(form Google Translate,^^), 
Is there a problem with my Mac?

zhoulilidamowang, 2017-05-08:
But my HID device in the win7 can be used normally(use hidapi)(form Google Translate,^^),
Is there a problem with my Mac?

signal11, 2017-05-08:
Don't know. Use your debugger and debug into the code. Also make sure you're using the latest from git (git clone ...)