signal11/hidapi #329: Cannot iterate through multiple devices of the same vendor and product ID.

Hobbyistpython, 2017-05-10:
Hello,

I am not able to use the serial number to identify a specific device connected to my computer, is this feature working?

Regards,

signal11, 2017-05-10:
Yes it is working.

Hobbyistpython, 2017-05-10:
Huh? Okay, must be something wrong with how it is stored. Is there another way I could easily select one of up to 6 individual usb devices all connected at the same time and sharing vendor and product ids?

tresf, 2017-05-11:
> Yes it is working.

Worth noting, there seems to be a bug in Windows where the serial number is returned as `?????`.

I have tried several HID libraries to no avail.

This issue is somewhat mentioned here: https://github.com/nyholku/purejavahidapi/issues/35

tresf, 2018-03-21:
In response to my previous message... the Windows serial number with question marks is specific to certain hardware, such as Mettler Toledo weight scales.  This doesn't occur for all HID hardware.

In regards to the original issue... @Hobbyistpython... if you're using Linux, for serial numbers to worka slight `udev` modification is needed as somewhat explained here, but without clear instructions for how to resolve it... #118.

Sample `udev` rule:

```bash
# usb override settings
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", MODE="0666"
```

This is a plain text file which can be placed (for example): `/lib/udev/rules.d/99-udev-override.rules`.

I believe the above technique is for `raw` mode only per @signal11's comment here: https://github.com/signal11/hidapi/issues/118#issuecomment-18018262.