signal11/hidapi #331: hid_write fails with EIO

opoplawski, 2017-05-22:
I've been using some TEMPer1V1.1 usb probes for a while now with RHEL6 with no (or at least very few) problems.  Just upgraded to RHEL7 and now hid_write() is failing with EIO.  I've tried commits 0a199c3 (what I had on EL6) though a6a622f but get the same result.  Kernel is 3.10.0-514.16.1.el7.x86_64.
```
open("/dev/hidraw1", O_RDWR)            = 3
ioctl(3, HIDIOCGRDESCSIZE or HIDIOCGVERSION or SNDRV_HWDEP_IOCTL_INFO, 0x7ffc0d8fa6ec) = 0
ioctl(3, HIDIOCAPPLICATION or HIDIOCGRDESC or SNDRV_HWDEP_IOCTL_DSP_STATUS, 0x7ffc0d8fa6f0) = 0
write(3, "\0\1\2003\1\0\0\0\0", 9)      = -1 EIO (Input/output error)
write(2, "/dev/hidraw1: Failed to read the"..., 67/dev/hidraw1: Failed to read the sensors: HID write failed: (null)
) = 67
close(3)                                = 0
```
The dev handle contains:
```
$6 = {device_handle = 7, blocking = 1, uses_numbered_reports = 0}
```

signal11, 2017-05-23:
So the write() system call fails. You can see the data being passed in. Are you sure your device can handle the report you're giving it (8-byte un-numbered report)?

You're going to need to do some debugging here. Maybe the old kernel was more forgiving. Are you sure you were using the hidraw implementation on RHEL-OLD with a 2.6.32-frankenkernel? hidraw didn't even really work well until 2.6.39. Look at linux/README.txt

opoplawski, 2017-05-23:
The same user space code ran okay on EL6 with its 2.6.32 series kernels.  Here's an strace from EL6:
```
open("/dev/hidraw2", O_RDWR)            = 3
ioctl(3, HIDIOCGRDESCSIZE or HIDIOCGVERSION or SNDRV_HWDEP_IOCTL_INFO, 0x7ffe1d27e914) = 0
ioctl(3, HIDIOCAPPLICATION or HIDIOCGRDESC or SNDRV_HWDEP_IOCTL_DSP_STATUS, 0x7ffe1d27e920) = 0
write(3, "\0\1\2003\1\0\0\0\0", 9)      = 9
poll([{fd=3, events=POLLIN}], 1, 1000)  = 1 ([{fd=3, revents=POLLIN}])
read(3, "\200\2\24p\320\37\305\275", 64) = 8
```
el6 dmesg:
```
input: RDing TEMPer1V1.1 as /devices/pci0000:00/0000:00:1d.0/usb6/6-2/6-2:1.0/input/input17
generic-usb 0003:0C45:7401.0003: input,hidraw1: USB HID v1.10 Keyboard [RDing TEMPer1V1.1] on usb-0000:00:1d.0-2/input0
generic-usb 0003:0C45:7401.0004: hiddev96,hidraw2: USB HID v1.10 Device [RDing TEMPer1V1.1] on usb-0000:00:1d.0-2/input1
```
el7 dmesg
```
[449164.170469] usb 2-1.2: Product: TEMPer1V1.1
[449164.174634] input: RDing TEMPer1V1.1 as /devices/pci0000:00/0000:00:1d.7/usb2/2-1/2-1.2/2-1.2:1.0/input/input34
[449164.225956] hid-generic 0003:0C45:7401.002D: input,hidraw4: USB HID v1.10 Keyboard [RDing TEMPer1V1.1] on usb-0000:00:1d.7-1.2/input0
[449164.228443] hid-generic 0003:0C45:7401.002E: hiddev0,hidraw5: USB HID v1.10 Device [RDing TEMPer1V1.1] on usb-0000:00:1d.7-1.2/input1
```

This is the code I'm using - https://github.com/edorfaus/TEMPered

Happy to do some more debugging, but really don't know where to go next.

However, just switched TEMPered to use the libusb back-end and that works.  So I guess this could be closed unless there is any more information that you'd like.

Quireg, 2017-07-27:
Since issue is still open I am wondering were you able to succeed with write() call to 3.10.0-514.16.1.el7.x86_64 kernel? I have faced exact same issue and not sure how to proceed further.

S1artie, 2018-07-19:
We have encountered a very similar issue as this one, also with a CentOS/RHEL 3.10.0-514 kernel (CentOS 7.3). It turned out that the write functionality works just fine on CentOS 7.2 and 7.4 kernels, so there had to be something special with the 7.3 series. I then started to perform code diffs of the USB HID device driver stack of these three versions and found the following:

Red Hat appears to have merged a bunch of commits into 7.3 that make up some apparently new functionality to use OUT interrupt endpoints to send data to USB HID devices from hidraw devices. As support for this appears to be optional according to the USB HID spec, they needed a fallback in case the device doesn't support such an endpoint. One of the commits they merged in - https://github.com/torvalds/linux/commit/3a75b24949a8d34d2014866006e30dc69dd567c8 - therefore contains exactly this fallback. The fallback looks for the error code value of ENOSYS, which means "Functionality not supported" - if that error code is thrown from the newer code path that tries to use the new feature for sending, it will fall back to the old code path. In case of any other error (or success) code, the code is immediately returned.

Unfortunately, Red Hat appeared to have left out another crucial commit that was actually performed 10 days earlier on the mainline kernel -  https://github.com/torvalds/linux/commit/ddea1af9d9387665ee194ae1d98ec0d49c169313 - in which the error code returned by the newer code path in case of non-availability of an OUT endpoint has been changed from EIO to ENOSYS. They only fixed this problem with CentOS 7.4, in which the missing commit has apparently been applied. All CentOS 7.3 kernels however don't use the fallback accordingly, but throw EIO errors instead.

This very much looks like a Red-Hat-manufactured kernel bug to me, so the best solution is to upgrade to a newer kernel that has the bug fixed (CentOS 7.4 and newer). Or to use the libusb HIDAPI backend, but while this works on CentOS 7.3 as well, it would break just as easily if Red Hat "decides" to wreck the libusb kernelspace code in one of their next minor releases.