signal11/hidapi #334: fixed : can't open hidapi device under Linux Mint 18 Cinnamon 64-bit

peterleimbach, 2017-06-04:
Hi signal11,

I installed your hidapi library with the testgui on a Linux Mint 18 Cinnamon 64-bit.

I could not connect with the testgui to the hidapi device.

The problem was that their was no working rule to change the ownership of the /dev/hidraw? device when the usb device is inserted. I checked this with "ls -l /dev/hidraw*".

I added in /etc/udev/rules.d/70-freestyle.rules a rule for my freestyle libre usb device.
> ACTION=="add|change", KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1a61", 
> ATTRS{idProduct}=="3650", TAG+="uaccess"

(Becarefully this is on line without leading > in the rules file.)

The numbers for idVendor und idProduct I got from the output from the "dmesg" command in the terminal.
> [ 1579.915612] usb 1-4: new full-speed USB device number 7 using xhci_hcd
> [ 1580.045254] usb 1-4: New USB device found, idVendor=1a61, idProduct=3650
> [ 1580.045261] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 1580.045265] usb 1-4: Product: ADC Reader
> [ 1580.045269] usb 1-4: Manufacturer: Abbott Diabetes Care
> [ 1580.048117] hid-generic 0003:1A61:3650.0006: hiddev0,hidraw3: USB HID v10.10 Device [Abbott 
> Diabetes Care ADC Reader] on usb-0000:00:14.0-4/input0

Now it works for me and maybe this helps others too.

Thanks for your work. Now I can read the data from my device in the linux box.

p.s. It's my first post on GIT. If I declare this issue as fixed in the topic and this is wrong sorry I would just like to help others and have to learn the issue process later.

best regards
Peter

flummoxedca, 2018-07-20:
Hi Peter, 

I found your comment after a search for 1a61:3650 usb linux. I just started using my Freestyle Libre and am interested in gathering data from it using Ubuntu. I am not a developer but use linux almost exclusively .

I'm curious if you have done more with your Libre and linux ?

Regards

Dave

peterleimbach, 2018-07-20:
Hi Dave,I am not using Linux for my Diabetes management. I only use Arch Linux as my Desktop environment beneath Mac OS X.For Diabetes Management I use Libre with Bluereader and xDrip on Andoid plus a Nightscout site at Heroku. In a few days I will complement this with a Dana RS insulin pump with Android APS to get a hyprid closed loop. It's fun and a huge step forward in T1 diabetes management.Best Regards Peter

peterleimbach, 2022-03-27:
Hi Dave,

no I have switched to Dexcom G6 where the data integration is made in another way.

Best Regards 
Peter

closed 2022-03-27 by peterleimbach